tasmota/tasmotizer #1: Tasmotizer.py not found Ubuntu 18.04.3 LTS

jtroberts1, 2019-12-04:
step 1. sudo apt install python3-pip
step 2. pip3 install tasmotizer
step3. run tasmotizer.py
  - Command not found

Did I miss a step?  I found tasmotizer in ~/.local/lib/python3.6/site-packages/

jtroberts1, 2019-12-04:
I got it to work.  Yeah, finally a simple tool in Linux to flash.  ESPtool would never let me do a full erase in Linux, now I can finally kill my Windows VM.  
Here's what I did.
- pip3 install PyQt5 pyserial (I don't think this is necessary)
- cd ~./local/bin
- sudo python3 tasmotizer.py
  - without sudo I get access denied to ttyUSB0

Works great.  I was able to flash my Etekcity ESW01 outlets.

Ty for the great tool

closed 2019-12-04 by jtroberts1

jziolkowski, 2019-12-05:
That's interesting, because I actually developed and tested that app on Kubuntu :) I (and a few users) tried the pip installation and it worked without a hitch.

Also, instead of using `sudo`, append your user to `dialout` group. After next login you'll have R/W access without sudo.

Glad you like the tool!