tasmota/tasmotizer #2: crashes after clicking Tasmotize!

mbravorus, 2019-12-05:
Installed with pip3 on a mac (OS X Mojave 10.14.6)

After setting the port and release binary crashes with

```
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 657, in start_process
    dlg = FlashingDialog(self)
  File "/usr/local/bin/tasmotizer.py", line 303, in __init__
    esptool.sw.read_start.connect(self.read_start)
AttributeError: module 'esptool' has no attribute 'sw'
Abort trap: 6
```

esptool.py is verified as working (I used it to flash after reproducing the crash several times)

pyserial verified to be v3.4

jziolkowski, 2019-12-05:
Can you uninstall esptool and check again? I suspect Python import the packaged esptool instead of the one modified provided by Tasmotizer.

mbravorus, 2019-12-06:
I'll try that on the next device I tasmotize (next week) and report

but realistically, most people who are going to install tasmotizer will have esptool already installed, so it might be a good idea to document this at a minimum, and maybe work around it somehow, in the general spirit of magic tasmotization :)

namadori, 2019-12-07:
I confirm your suspect, uninstalling esptool and reinstalling tasmotizer avoided the error.

jziolkowski, 2019-12-07:
Thank you. I'll have to add it to 1.1 so it forces loading of the current .py file.

mbravorus, 2019-12-08:
I also confirm that reinstallation after removing esptool works as described.

It's not very convenient though (because I sometimes need just the esptool, say, to quickly do a read_mac or something), so new solution in 1.1 will be great! Current workaround is to keep tasmotizer (or esptool) in a virtualenv.

jziolkowski, 2019-12-08:
I've created a new branch (dev-1.1) with renamed esptool to enforce local import if esptool is present in python installation. Please test.

jziolkowski, 2019-12-09:
the changes are now merged and released.

closed 2019-12-09 by jziolkowski