tasmota/tasmotizer #4: tasmotizer crashes after first operation

mbravorus, 2019-12-08:
This is an issue when working with some devices which cannot self-reset successfully (e.g. Sonoff Touch switches). After *each* operation (original firmware backup, erase, flash), the device needs to be manually reset by power-cycling and bringing GPI00 to GND on boot. It is generally not possible to do while tasmotizer (underlying esptool) waits for connection, the timeout is not long enough and the app crashes instead of handling that gracefully. Also, if it involves power-cycling the usb dongle, the device gets destroyed/recreated and the app crashes again.

example:

```
$ tasmotizer.py 
esptool.py v2.8
Serial port /dev/tty.usbserial-1412210
Connecting....
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 2c:f4:32:aa:7d:57
Uploading stub...
Running stub...
Stub running...
Read 1048576 bytes at 0x0 in 95.7 seconds (87.7 kbit/s)...
Hard resetting via RTS pin...
esptool.py v2.8
Serial port /dev/tty.usbserial-1412210
Connecting........_____
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 493, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 378, in command
    p = self.read()
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 323, in read
    return next(self._slip_reader)
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 2072, in slip_reader
    read_bytes = port.read(1 if waiting == 0 else waiting)
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 323, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (6, 'Device not configured')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 86, in execute
    esptool.main(command)
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 2931, in main
    esp.connect(args.before)
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 506, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 486, in _connect_attempt
    self.sync()
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 427, in sync
    self.command()
  File "/usr/local/lib/python3.7/site-packages/esptool.py", line 389, in command
    self._port.timeout = saved_timeout
  File "/usr/local/lib/python3.7/site-packages/serial/serialutil.py", line 368, in timeout
    self._reconfigure_port()
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (6, 'Device not configured')
Abort trap: 6

```

Also, at least on mac os, all tasmotizer exceptions are critical and bring up big system stack trace report windows, maybe it doesn't have to be so severe? Looks like this:

System stack trace report - https://prnt.sc/q7uzuv

I'm not entirely sure it is possible, but an option to wait between operations (so that any manual connection/resets could be done) could solve this.

jziolkowski, 2019-12-08:
notice that you are using the esptool from your python installation, not from tasmotizer. I've noticed that error in a separate issue. Please uninnstall esptool from your system and try again.

I've used tasmotizer on non self-resetting devices and it worked fine.

jziolkowski, 2019-12-08:
Also, as long as the device stays in bootloader mode (and it should, because it doesn't reset automatically) all steps can be done one after another, no need to reset in between

mbravorus, 2019-12-08:
I actually did remove esptool prior to that run:

```
hylozoist:tasmota mbravo$ pip3 uninstall esptool
Uninstalling esptool-2.8:
  Would remove:
    /usr/local/bin/espefuse.py
    /usr/local/bin/espsecure.py
    /usr/local/bin/esptool.py
    /usr/local/lib/python3.7/site-packages/espefuse.py
    /usr/local/lib/python3.7/site-packages/espsecure.py
    /usr/local/lib/python3.7/site-packages/esptool-2.8.dist-info/*
    /usr/local/lib/python3.7/site-packages/esptool.py
Proceed (y/n)? y
  Successfully uninstalled esptool-2.8
hylozoist:tasmota mbravo$ pip3 install tasmotizer
Requirement already satisfied: tasmotizer in /usr/local/lib/python3.7/site-packages (1.0)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/site-packages (from tasmotizer) (3.4)
Requirement already satisfied: PyQt5>=5.10 in /usr/local/lib/python3.7/site-packages (from tasmotizer) (5.13.2)
Requirement already satisfied: PyQt5_sip<13,>=4.19.19 in /usr/local/lib/python3.7/site-packages (from PyQt5>=5.10->tasmotizer) (12.7.0)
hylozoist:tasmota mbravo$ pip3 uninstall tasmotizer
Uninstalling tasmotizer-1.0:
  Would remove:
    /usr/local/bin/tasmotizer.py
    /usr/local/lib/python3.7/site-packages/banner.py
    /usr/local/lib/python3.7/site-packages/gui.py
    /usr/local/lib/python3.7/site-packages/tasmotizer-1.0.dist-info/*
    /usr/local/lib/python3.7/site-packages/tasmotizer.py
Proceed (y/n)? y
  Successfully uninstalled tasmotizer-1.0
hylozoist:tasmota mbravo$ pip3 install tasmotizer
Collecting tasmotizer
  Using cached https://files.pythonhosted.org/packages/0d/ba/bf4e67e90f3fb1c25405b640af8073447bbc1b60268a8cdec4156bd46723/tasmotizer-1.0-py3-none-any.whl
Requirement already satisfied: PyQt5>=5.10 in /usr/local/lib/python3.7/site-packages (from tasmotizer) (5.13.2)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/site-packages (from tasmotizer) (3.4)
Requirement already satisfied: PyQt5_sip<13,>=4.19.19 in /usr/local/lib/python3.7/site-packages (from PyQt5>=5.10->tasmotizer) (12.7.0)
Installing collected packages: tasmotizer
Successfully installed tasmotizer-1.0
hylozoist:tasmota mbravo$ esptool.py -p /dev/tty.usbserial-1412210 read_mac
-bash: /usr/local/bin/esptool.py: No such file or directory
```
I'll check if I missed something and it is for some reason still there... but not sure how it could happen?

reset comes from esptool command where it says `Hard resetting via RTS pin...` after the operation, if we can skip that, the problem would disappear, I agree

jziolkowski, 2019-12-08:
`Hard resetting via RTS pin...` is only a command sent from esptool to the port adapter. If the RTS isn't connected to the device (and it's not, because you only connect GPIO0 with GND, and RX/TX from the adapter) then the device is NOT being reset. This is why devices like NodeMCU etc can reset - because they have more pins connected from the USB port. In case of devices when you connect only TX/RX this is not the case. The device is kept in bootloader state for as long as you will disconnect it.

Jason2866, 2019-12-16:
Esptool.py  is not needed to deinstall anymore with release 1.1

closed 2019-12-16 by Jason2866

b0gda0, 2020-06-05:
Hi. I'm new to this and I want to flash tasmota on a sonoff basic. The switch is the older model without the Black cables running on top. When I try to flash with backup option it start runnig the backup and it freezes at 100% after the backup is finished. It doesn't start the flash of tasmota. I waited more than 5 min and nothing happenes. The only option i have is to hit abort button and the process is cancel. Thank you.

JayJay2105, 2020-06-05:
> Hi. I'm new to this and I want to flash tasmota on a sonoff basic. The switch is the older model without the Black cables running on top. When I try to flash with backup option it start runnig the [...]

The backup file has been created, check in your file explorer, then just flash again without the backup option

b0gda0, 2020-06-06:
hi,
thank you for your reply and advice. i tried flashing without running the backup but unfortunately it doesn't work. i have the ch340g USB to serial connector and i have read that when you select the 3.3v voltage it actually uses 3.6v or 3.9v on the rx and tx and this is not ok with sonoff switch i want to flash. so i'll try buying another usb to serial connector and i'll try again. thanks.