tasmota/tasmotizer #5: Install in Linux mint PyQt5 Issue-Windows only ?

8art-5imson, 2019-12-08:
Can't get running in linux mint. pip3 install PyQt5 pyserial returns >
  Could not find a version that satisfies the requirement PyQt5 (from versions: )
No matching distribution found for PyQt5.
Is PyQt5 windows only ? if so what can be used instead.

jziolkowski, 2019-12-08:
Which python version? PyQT5 is available on all platforms for python3

jziolkowski, 2019-12-08:
also you can do `pip3 install tasmotizer` and it should also pull the requirements for you

8art-5imson, 2019-12-08:
chaos@chaos-tech:~$ python --version
Python 3.6.9

chaos@chaos-tech:~$ pip3 install tasmotizer
Collecting tasmotizer
  Using cached https://files.pythonhosted.org/packages/0d/ba/bf4e67e90f3fb1c25405b640af8073447bbc1b60268a8cdec4156bd46723/tasmotizer-1.0-py3-none-any.whl
Collecting pyserial>=3.0 (from tasmotizer)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting PyQt5>=5.10 (from tasmotizer)
  Could not find a version that satisfies the requirement PyQt5>=5.10 (from tasmotizer) (from versions: )
No matching distribution found for PyQt5>=5.10 (from tasmotizer)

jziolkowski, 2019-12-08:
try `python3 -m pip install PyQt5`

jziolkowski, 2019-12-08:
or the same with tasmotizer

8art-5imson, 2019-12-08:
chaos@chaos-tech:~$ python3 -m pip install PyQt5
Collecting PyQt5
  Could not find a version that satisfies the requirement PyQt5 (from versions: )
No matching distribution found for PyQt5

jziolkowski, 2019-12-08:
This is most interesting. I'm using Python 3.6.8 as my dev platform and it installs without a hitch.
you could go to pypi PyQt5 and download the .whl file and install it manually

8art-5imson, 2019-12-08:
i have played around with this for a couple of hours now, I will delete all and start again, will try the .whl thing first. Where do i put the whl file ?

jziolkowski, 2019-12-08:
wherever you want. just install it using `pip install <filename>.whl`
I also suggest you do all that in a clean venv

8art-5imson, 2019-12-08:
Still no Go...
chaos@chaos-tech:/tasmotize/venv/tasmotizer$ sudo -H pip3 install PyQt5 pyserial
Collecting PyQt5
  Could not find a version that satisfies the requirement PyQt5 (from versions: )
No matching distribution found for PyQt5

jziolkowski, 2019-12-08:
Then I'm sorry, but this is out of support scope of this project. Before releasing, I and the team have done pip tests on around dozen test releases of tasmotizer, on a variety of OSes, and this issue has not been encountered. Please seek advise on a forum related to your OS.

8art-5imson, 2019-12-08:
Thanks for all the help anyway.

jziolkowski, 2019-12-08:
BTW did you try with manual download/install of the .whl file?

closed 2019-12-09 by jziolkowski