tasmota/tasmotizer #6: No module named 'tasmotizer_esptool'

robin13, 2019-12-09:
I installed with `sudo pip3 install tasmotizer ` and then ran:

```
➜ ~ tasmotizer.py                            
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 7, in <module>
    import tasmotizer_esptool as esptool
ModuleNotFoundError: No module named 'tasmotizer_esptool'
```

jziolkowski, 2019-12-09:
Dang, wrong dash :D I'll send a new package to pypi, give me a few minutes.

jziolkowski, 2019-12-09:
Uploaded tasmotizer 1.1.1 to pypi, the index should rebuild in a few minutes

jziolkowski, 2019-12-09:
@Jason2866 no need to rebuild anything.

closed 2019-12-16 by jziolkowski

jeylites, 2020-09-30:
I'm having the same problem. 

```
 File "/usr/local/bin/tasmotizer.py", line 7, in <module>
    import tasmotizer_esptool as esptool
```

jeylites, 2020-09-30:
Just updated Tasmotizer, seem to be working now.

dangerweenie, 2024-05-24:
just installed on ubuntu, having the same issue:

ModuleNotFoundError: No module named 'tasmotizer_esptool'