tasmota/tasmotizer #7: Needs improvements

mkh595, 2019-12-15:
It looks that this is easy tool to flash firmware and not need to deal with long command line or addresses where to put which file (ESP8266flasher).

But program needs improvements to avoid crashes (tested windows version)

- when selected port is already open e.g. in serial monitor, program crash. It should say that serial port cannot be open and allow to select another port or start flash again after port is closed

I tested with Wemos D1 mini

- own selected bin file was uploaded correctly

- upload release option ended with error

[image]

- after previous error I am not able to flash back neither my own file. 

[image]

- after sucessfull flash device cannot be flashed again until is physically disconnected. Same error like above

- when button OK is pressed it crash instead of stay open

jziolkowski, 2019-12-15:
It's not possible to kill a device by a bad flash. As for the remaining issues I'll have a look.

jziolkowski, 2020-01-10:
The device can't be programmed again after flashing because Tasmotizer stick to the default esptool behavior of resetting the device after flashing, ergo triggering a normal boot. I'll add that as an option for the user in the next release, including the check if the port is already claimed.

As for the reasons of why other software is also unable to flash your device: there can be a myriad reasons, starting with the faulty device, faulty flash, faulty adapter etc etc, so I can't really tell anything there.

jziolkowski, 2020-02-10:
Closing due to lack of feedback.

closed 2020-02-10 by jziolkowski