tasmota/tasmotizer #9: Request

titomontes, 2019-12-17:
Finally, a friendly GUI!

The program seems to work well, but i think there may be room for improvement in the config part. It could allow for sending a static IP setting, which would save the need to look for one in the router. Add that and we would have an instant config and known IP.

Thanks for making tinkering easier

borpin, 2020-01-13:
Yes I will +1

Being able to set Network settings (IP, Mask, gateway, DNS) would be a real bonus.

Option to set HA discoverable.

RaSTuS26, 2020-02-25:
Definitely a +1 here, static IP is a must for me, don't use DHCP at all.

jziolkowski, 2020-04-13:
Added in the development version (it will require manual installation of the required packages)

jziolkowski, 2020-09-21:
develop-1.3 containts the new send config dialog, and all the fixes and improvements from newly released 1.2, please test if possible