tasmota/tasmotizer #10: Windows Defender and AVG report Windows executable infected with a virus

netcompsys, 2019-12-23:
[image]

If you want to use this you have to disable Antivirus otherwise it removes it automatically

jziolkowski, 2019-12-23:
As it is written on the main page of the repo, it's a known false positive. Add it to whitelist and it's fixed.

closed 2019-12-23 by jziolkowski

netcompsys, 2019-12-23:
It is still an issue, making a comment and closing it does not fix the problem

jziolkowski, 2019-12-23:
False positive is a problem of antivirus, not ours. Or at least, it's pyinstaller problem, because this is what is used for packaging to .exe

Googling that error shows that antivirus companies give little attention to such reports and raised issues. So it's a bit of a ping-pong game of who should fix that. And when an antivirus mistakenly treats some code as malicious, it's not our responsibility to fix their problem.