tasmota/tasmotizer #11: How to send config

rebel7580, 2019-12-28:
I programmed a device successfully, then tried to use the "Send Config" option.  What I got was a grayed out screen with no way to enter any info to it. I suppose I'm doing something wrong, but not sure what.
[image]

[image]

With this screen when I click "save" it attempts to upload the default config to the device.

jziolkowski, 2019-12-28:
You must first click the box near the section you want to configure. Then that section will be enabled for input.

rebel7580, 2019-12-28:
Ah. With that dark theme, I didn't even see the check boxes! :)
Thanks.

closed 2019-12-28 by rebel7580