tasmota/tasmotizer #12: fails to start on linux mint 19.3

mprins, 2019-12-29:
It seems your requirements are out-of-date or incomplete

```
sudo apt install python3-serial python3-pyqt5 --reinstall 
...
Setting up python3-serial (3.4-2) ...
Setting up python3-pyqt5 (5.10.1+dfsg-1ubuntu2) ...
... 
git clone https://github.com/tasmota/tasmotizer.git
cd tasmotizer
python3 tasmotizer.py

Traceback (most recent call last):
  File "tasmotizer.py", line 15, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'

```

I also tried:

```
python3 -m venv ./tasmotizer-venv
source tasmotizer-venv/bin/activate
pip install --upgrade pip setuptools pyserial --upgrade
pip install tasmotizer
Collecting tasmotizer
  Using cached https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl
Collecting PyQt5>=5.10 (from tasmotizer)
  Using cached https://files.pythonhosted.org/packages/7c/5b/e760ec4f868cb77cee45b4554bf15d3fe6972176e89c4e3faac941213694/PyQt5-5.14.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-z9oy7374/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-z9oy7374/PyQt5/

```

jziolkowski, 2019-12-29:
QSerialPort is available since Qt5.1, therefore it should work when you installed python3-PyQt5 version 5.10. As for the missing module import error, you didn't notice that qserialport is packages as a separate .deb and needs to be installed manually. However, I'm not using python with system-wide packages, so I haven't tested if it works correctly.

The venv method using pip installed PyQt5 5.14 as you can see, ergo all the requirements are fulfilled.

However, for some odd reason, your pip command pulls the .tar.gz source and tries to install it, which is obviously wrong.

I've tried what you did, and in my case the whole venv installed properly:
```
faza@shrike:~/Projects$ python3 -m venv vtas
faza@shrike:~/Projects$ cd vtas/
faza@shrike:~/Projects/vtas$ source bin/activate
(vtas) faza@shrike:~/Projects/vtas$ pip install --upgrade pip setuptools pyserial --upgrade
Cache entry deserialization failed, entry ignored

Collecting pip
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/00/b6/9cfa56b4081ad13874b0c6f96af8ce16cfbc1cb06bedf8e9164ce5551ec1/pip-19.3.1-py2.py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 801kB/s 
Cache entry deserialization failed, entry ignored
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/54/28/c45d8b54c1339f9644b87663945e54a8503cfef59cf0f65b3ff5dd17cf64/setuptools-42.0.2-py2.py3-none-any.whl (583kB)
    100% |████████████████████████████████| 583kB 2.0MB/s 
Collecting pyserial
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Installing collected packages: pip, setuptools, pyserial
  Found existing installation: pip 9.0.1
    Uninstalling pip-9.0.1:
      Successfully uninstalled pip-9.0.1
  Found existing installation: setuptools 39.0.1
    Uninstalling setuptools-39.0.1:
      Successfully uninstalled setuptools-39.0.1
Successfully installed pip-19.3.1 pyserial-3.4 setuptools-42.0.2
(vtas) faza@shrike:~/Projects/vtas$ 
(vtas) faza@shrike:~/Projects/vtas$ pip install tasmotizer
Collecting tasmotizer
  Downloading https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl (243kB)
     |████████████████████████████████| 245kB 283kB/s 
Collecting PyQt5>=5.10
  Downloading https://files.pythonhosted.org/packages/88/d0/4572492d633e311c0143ba20596cb5c090d37442d1cd431082193f3bbb3a/PyQt5-5.14.0-5.14.0-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (63.5MB)
     |████████████████████████████████| 63.5MB 25.1MB/s 
Requirement already satisfied: pyserial>=3.0 in ./lib/python3.6/site-packages (from tasmotizer) (3.4)
Collecting PyQt5-sip<13,>=12.7
  Using cached https://files.pythonhosted.org/packages/69/d2/92d85b0865fe68755488a30eec1c2b1b14da6c96c74441dc1c88e0659397/PyQt5_sip-12.7.0-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: PyQt5-sip, PyQt5, tasmotizer
Successfully installed PyQt5-5.14.0 PyQt5-sip-12.7.0 tasmotizer-1.1.1
```
And tasmotizer starts as excepted.

This error occurs somewhat often with pip, and as such, is outside of the scope of support for this application.

stelford, 2020-01-06:
alternativaly... sudo apt-get install python3-pyqt5.qtserialport

jaydub385, 2020-01-09:
sane issue on unbuntu 19: ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'

sudo apt install python3-serial python3-pyqt5 --reinstall 

that fixed it!

closed 2020-01-10 by jziolkowski

ouinouin, 2020-09-16:
under ubuntu focal : needs also this package to be able to start : python3-pyqt5.qtserialport

muenznej, 2020-12-13:
> alternativaly... sudo apt-get install python3-pyqt5.qtserialport

i looked through all these "solutions" and this easy one solved it with ease. thank you