tasmota/tasmotizer #14: MacOS 10.12.6: Import error, Symbol not found: _futimens

CaptClaude, 2020-01-03:
First try to install on a Macbook Pro, OS X 10.12.6, installs correctly but will not run.
Is this an "Update Your Operating System" and forward compatibility issue?  
```
Referenced from: /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.13)
```
All the messages from install to fail:
```
jm:Tasmota jm$ pip3 install tasmotizer
Collecting tasmotizer
  Downloading https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl (243kB)
     |████████████████████████████████| 245kB 1.3MB/s 
Collecting PyQt5>=5.10
  Downloading https://files.pythonhosted.org/packages/1e/8f/dd8312af6ed44c15419fb14c38b6532eeb765c428f0642e30dc3ef26b6d4/PyQt5-5.14.0-5.14.0-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (43.0MB)
     |████████████████████████████████| 43.0MB 2.4MB/s 
Requirement already satisfied: pyserial>=3.0 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from tasmotizer) (3.4)
Collecting PyQt5-sip<13,>=12.7
  Downloading https://files.pythonhosted.org/packages/91/13/306b26bc8870c63e30af47d18a5dee5253adf5b523b2d0afc1fc023affdc/PyQt5_sip-12.7.0-cp36-cp36m-macosx_10_6_intel.whl (124kB)
     |████████████████████████████████| 133kB 387kB/s 
Installing collected packages: PyQt5-sip, PyQt5, tasmotizer
  Found existing installation: PyQt5 5.9
    Uninstalling PyQt5-5.9:
      Successfully uninstalled PyQt5-5.9
Successfully installed PyQt5-5.14.0 PyQt5-sip-12.7.0 tasmotizer-1.1.1
jm:Tasmota jm$ tasmotizer.py
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/tasmotizer.py", line 7, in <module>
    import tasmotizer_esptool as esptool
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tasmotizer_esptool.py", line 36, in <module>
    from PyQt5.QtCore import pyqtSignal, QObject, pyqtSlot
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/PyQt5/QtCore.abi3.so, 2): Symbol not found: _futimens
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore
```
Thanks.

jziolkowski, 2020-01-10:
As I don't have a Mac, please visit the official tasmota discord server (https://discord.gg/Ks2Kzd4) and ask there, or seek help for PyQt5 and MacOS on google, as I can't help in this regard.

closed 2020-01-10 by jziolkowski

CaptClaude, 2020-01-11:
Tried Dischord on 1/3 -- no luck.  Nothing from Google.  Probably an OS issue. Will try the PyQt5 folks.