tasmota/tasmotizer #15: Error 2

vav4eg, 2020-01-09:
Ubuntu 18.04.3

> 
> vav4eg@vav4eg:~$ git clone https://github.com/tasmota/tasmotizer.git
> Клонирование в «tasmotizer»…
> remote: Enumerating objects: 89, done.
> remote: Counting objects: 100% (89/89), done.
> remote: Compressing objects: 100% (64/64), done.
> remote: Total 89 (delta 43), reused 42 (delta 19), pack-reused 0
> Распаковка объектов: 100% (89/89), готово.
> vav4eg@vav4eg:~$ cd ^C
> vav4eg@vav4eg:~$ cd tasmotizer
> vav4eg@vav4eg:~/tasmotizer$ pip3 install tasmotizer
> Collecting tasmotizer
>   Using cached https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl
> Collecting pyserial>=3.0 (from tasmotizer)
>   Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
> Collecting PyQt5>=5.10 (from tasmotizer)
>   Using cached https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz
>     Complete output from command python setup.py egg_info:
>     Traceback (most recent call last):
>       File "<string>", line 1, in <module>
>       File "/usr/lib/python3.6/tokenize.py", line 452, in open
>         buffer = _builtin_open(filename, 'rb')
>     FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-8any1nz_/PyQt5/setup.py'
>     
>     ----------------------------------------
> Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-8any1nz_/PyQt5/

Smitkas, 2020-01-09:
same trouble
pi@raspberrypi:~ $ pip3 install tasmotizer 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting tasmotizer
  Using cached https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl
Collecting PyQt5>=5.10 (from tasmotizer)
  Using cached https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-q2o2i3s0/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-q2o2i3s0/PyQt5/

jziolkowski, 2020-01-10:
This is related to pip and PyQt5 not tasmotizer. For some reason, pip tries to install the source code instead of the distribution files of PyQt5.

closed 2020-01-10 by jziolkowski

vav4eg, 2020-01-12:
The problem was resolved after
```
hash -d pip
hash -r pip
```

dhavaldesai-HA, 2020-01-12:
Just for the info to future users:

I had following error while trying to install Tasmotizer on Ubuntu 18.04

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-w9qn5pg_/PyQt5/

Following solved the problem ::
$ sudo pip3 install --upgrade pip setuptools wheel

Sorry if it sounds silly to expert users. Just trying to help to newbies like me ....

jziolkowski, 2020-01-12:
Thanks, I'll add that to the readme file.

DirkTas67, 2020-01-25:
> Just for the info to future users:
>
> I had following error while trying to install Tasmotizer on Ubuntu 18.04
>
> Command "python setup.py egg_info" failed with error code 1 in [...]

thank you!!!

L4RGER, 2020-08-13:
The above actually made my situation worse, it took me a couple of days to find the solution. I could install due to PyQt5 not installing correctly which lead me to [this thread where I posted the solution I found](https://github.com/tasmota/tasmotizer/issues/26). It did take a couple of hours though.