tasmota/tasmotizer #20: pip install fails on recent ubuntu/debian releases due to PyQt5

cjd, 2020-01-20:
When trying to install on recent debian/ubuntu it fails to build PyQt5
===
$ pip3 install --user tasmotizer
Collecting tasmotizer
  Downloading https://files.pythonhosted.org/packages/2d/eb/def84f3c8667dde54e7c732b82e657cd2b7313e7a6d8009b79338bb278cc/tasmotizer-1.1.1-py3-none-any.whl (243kB)
    100% |████████████████████████████████| 245kB 2.1MB/s 
Collecting PyQt5>=5.10 (from tasmotizer)
  Downloading https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz (3.2MB)
    100% |████████████████████████████████| 3.2MB 343kB/s 
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-9r2xzzk3/PyQt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-9r2xzzk3/PyQt5/
===

As a workaround you can install the system package instead via "sudo apt-get install python3-pyqt5"
Then the pip install completes

helgeschneider, 2020-01-21:
i had different issues with debian buster and ubuntu 16.05
Fixed them in buster with:

apt install python3-pip python3-serial python3-pyqt5 --reinstall
pip3 install --upgrade pip
pip install tasmotizer

jziolkowski, 2020-01-21:
See https://github.com/tasmota/tasmotizer/issues/15#issuecomment-573445180

closed 2020-01-21 by jziolkowski