tasmota/tasmotizer #22: Feature Request: OTA

fmarzocca, 2020-02-02:
It would be nice if Tasmotizer could also be used OTA, by detecting any OTA-enabled device on the LAN and list that in the "Port" section.

DaleChaney, 2020-02-10:
I would really like an OTA option, because I can't do OTA the way Itead says,  I've destroyed 7 mini's and counting, because of their way of reloading a mini ota.  This would be a Godsend if you could.

austwhite, 2020-04-25:
Also would like this feature.  Tasmotizer is great, but sometimes it would be nice to be able to use it to flash bin files OTA, especially on some smart switches which are difficult to disassemble and solder wires to.
(assumption - smart switches have already been converted using Tuya Convert or similar process)

Jason2866, 2020-08-10:
There will be no Tuya Convert features. Tuya Convert uses a exploit. 
Sooner or later this security hole will be closed. Invested time is a 100% loss.
If you want to Tuya Convert use the original which works nice and is well maintained

DaleChaney, 2020-08-10:
Sent from Mail for Windows 10 From: Jason2866Sent: Monday, August 10, 2020 9:58 AMTo: tasmota/tasmotizerCc: DaleChaney; CommentSubject: Re: [tasmota/tasmotizer] Feature Request: OTA (#22) There will be no Tuya Convert features. Tuya Convert uses a exploit.Sooner or later this security hole will be closed. Invested time is a 100% loss.If you want to Tuya Convert use the original which works nice and is well maintained—Jason….  I know Tuya will close the hole eventually, but once flashed, the switches continue until you must make a change.  That’s better than what I have with Sonoff. I’ve been using Tuya-convert 2.3 for a while now with no issues.  I’m going to flash 20-30 switches in the next day or so.Thanks for your comments.  Dale

Jason2866, 2020-08-29:
Will not implemented

closed 2020-08-29 by Jason2866

STManiion, 2020-12-22:
Wish I'd read this exchange B4 installing the thing. Other outfits will do both. Also unimpressed by several people politely expressing their wishes and being shut down in such a rude, preemptive manner. BTW apparently Tuya has updated, if this flood isn't from an earlier up[date. A billion pieces of the now "obsolete" models, the ones we want, are flooding the market at next to nothing. if you want to wait a while the Chinese  wholesalers will sell you more than you could ever use for pocket change.

Jason2866, 2020-12-22:
@STManiion  It is OpenSource. If you want to see the feature do a fork and implement this feature.
Since Tuya Conbert uses a Security Hole (which is closed already) implementing Tuya Convert would waste our rare time.
Keep in mind everyone is doing coding in free time.

jziolkowski, 2020-12-22:
Best I can do I think is to add "plain" OTA server, without tuya support

jziolkowski, 2020-12-22:
I mean, if it works with tuya then fine, but it will not be the goal. Tuya-convert process is best made using their dedicated toolchain

jziolkowski, 2020-12-22:
I'm creating a feature request in project discussion board for OTA server, please comment there if there's anything to add related to the request.

STManiion, 2020-12-23:
Wow. I'm pretty new here, didn't expect anyone to even notice my comments. Thanks, I'll look for that.