tasmota/tasmotizer #23: OTA option would make this very valuable for MINI's that can't be flashed

DaleChaney, 2020-02-10:
Mini's can be flashed OTA, unless you're one of the many people having problems with it.  Like me, many are having failures trying,  I've bricked 7 already.  OTA is where Itead is going (already got 3 of them).  
Do you have any idea how many people would love this?

jziolkowski, 2020-02-10:
You mean the OTA server, ot the whole OTA flashing process?

hristo-atanasov, 2020-02-11:
@DaleChaney the pieces, that you have bricked, can be flashed using the good old method - with serial connection + Serial adapter. There are pads left for that on the MINI's board. I still haven't used OTA at all, as I use the serial connection always. :) The plastic enclosure/shell of the mini's can be opened with bare hands. Using this method and investing some time (as some soldering skills are needed to solder a wire to the pads) will bring them to life .. ;) And yes .. If there can be easier way, I would appreciate it .. And use it, too .. :)

Jason2866, 2020-08-10:
If sonoff would change the OTA DIY process, there would be a chance for.
How it works (or not) at the moment is is a mess...

DaleChaney, 2020-08-10:
Sent from Mail for Windows 10 From: Jason2866Sent: Monday, August 10, 2020 10:06 AMTo: tasmota/tasmotizerCc: DaleChaney; MentionSubject: Re: [tasmota/tasmotizer] OTA option would make this very valuable for MINI's that can't be flashed (#23) If sonoff would change the OTA DIY process, there would be a chance for.How it works (or not) at the moment is is a mess...—Ewelink and support for the Mini is more than a mess.  It’s a true disaster.  The reviews under Ewelink are a comedy.  90% of the reviewers can get it working, and often, they can’t even pair their switches.  Out of the most recent 20-30 of the reviews, they say it hasn’t worked since early in July.  I’ve been waiting and trying for over a year, with no results.  The OTA flashing is a joke.  I’ve been telling them that mDNS does not work on my systems, probably because I don’t use iTunes (which is where Bonjour comes from).  And, if mDNS doesn’t work, then Ewelink cannot process my switches.  Very frustrating.I need to do OTA because I am handicapped – One usable arm and one usable leg.  So, soldering is simply not an option for me.  I bought alligator clip jigs to hold the wires and boards, but it’s simply not doable for me.  Why such small pads on the Mini?  Why aren’t they through drilled?  Perhaps the don’t want us to Tasmotize them? I’m so upset with Sonoff, that I’ve decided to go with Tuya switches instead (after purchasing 20 Sonoff Mini’s that don’t work at all).  The Tuya Mini switches, like the Sonoff Mini’s, also hide inside electical boxes, and theirv OTA process just WORKs..  And, they have the same spec’s.  So, I’ll just pay a couple bucks more, but with less frustration. One would think, the company that owned this Smart business from the beginning, would want to capitalize on it.  How foolish.

Jason2866, 2020-08-10:
@DaleChaney  Take a look at Shelly devices no soldering needed for flashing.
With https://github.com/yaourdt/mgos-to-tasmota you can flash many Shelly devices OTA with Tasmota. Shelly devices have a  better circuit designed than sonoff minis...

Jason2866, 2020-08-29:
Will not implemented

closed 2020-08-29 by Jason2866