tasmota/tasmotizer #25: REQUEST: Show new ip address

SergioEscuder, 2020-03-07:
feature request. once wifi config info has been uploaded, read the serial output and show in tasmotizer the IP the device got.

ionciubotaru, 2020-03-08:
One more sugestion, please add a new tab to display serial output and serial input (similar to Arduino IDE). It is useful to verify if ESP succesfull conect to wifi or to send more commands.

jziolkowski, 2020-03-10:
In my current dev version (not yet available) I've added complete IP settings to Tasmotizer; I'll add a status bar so it will also show the detected IP after device reboot. Stay tuned.

ionciubotaru, 2020-03-10:
Good news.
Please add console to run:
backlog ssid1 abc;password1 cdf;mqtthost 192.168.x.y;ssid2 xyx;password2 fff ......

Jason2866, 2020-03-12:
Not the use case of Tasmotizer. For Tasmota device management there is TDM:
https://github.com/jziolkowski/tdm

JayJay2105, 2020-03-31:
Any idea on timescales for the version with ip address in, I'm new to all home automation, trying to learn Home Assistant and use Sonoff and H801 devices with Tasmota, my 1st attempt is with a H801 but I'm not getting an IP address for it so am not sure I'm doing correctly, and having this would help...
If you have any guidance this is what I have done,
Flashed using windows Tatomizer, full bin file, backed up old - get a crash when I click ok at the end.
re-flashed without backup, removed jumper and reset power, the added config, just wifi, re-powered on, and h801 signal LED flashes, but I don't get any new devices in my router ip list. have also tried a scan with TasmoAdmin.
Great work from everything I've seen, tasmota and home assistant have made me take the leap into trying to learn all this!

JayJay2105, 2020-04-01:
> Any idea on timescales for the version with ip address in, I'm new to all home automation, trying to learn Home Assistant and use Sonoff and H801 devices with Tasmota, my 1st attempt is with a H801 [...]

Fixed my problem! yay... but would be great to have IP shown...

jziolkowski, 2020-04-13:
This feature is added in the development version (it will require manual installation of the required packages)

scottchiefbaker, 2020-04-29:
This sounds like a great feature. Can't wait to see it.

jziolkowski, 2020-09-20:
"Get IP" available for testing in 1.1d branch, to be released tomorrow if all goes well

Jason2866, 2020-09-21:
Feature "Get IP" added in v.1.2

closed 2020-09-21 by Jason2866