tasmota/tasmotizer #26: Installation failure Raspbian buster.

SamuelCarson, 2020-03-07:
pi@raspberrypi:~ $ pip3 install tasmotizer
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting tasmotizer
  Using cached tasmotizer-1.1.1-py3-none-any.whl (243 kB)
Collecting PyQt5>=5.10
  Using cached PyQt5-5.14.1.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpgsh433v9
         cwd: /tmp/pip-install-z0ot5wcg/PyQt5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 103, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 257, in <module>
        main()
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 240, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 108, in prepare_metadata_for_build_wheel
        config_settings)
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 136, in _get_wheel_metadata_from_wheel
        whl_basename = backend.build_wheel(metadata_directory, config_settings)
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/sipbuild/api.py", line 51, in build_wheel
        project = AbstractProject.bootstrap('pep517')
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/sipbuild/abstract_project.py", line 82, in bootstrap
        project.setup(pyproject, tool, tool_description)
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 387, in setup
        self.apply_user_defaults(tool)
      File "project.py", line 62, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/pyqtbuild/project.py", line 86, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 202, in apply_user_defaults
        self.builder.apply_user_defaults(tool)
      File "/tmp/pip-build-env-ssz2bvja/overlay/lib/python3.7/site-packages/pyqtbuild/builder.py", line 68, in apply_user_defaults
        "specify a working qmake or add it to PATH")
    sipbuild.pyproject.PyProjectOptionException
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpgsh433v9 Check the logs for full command output.
pi@raspberrypi:~ $

SamuelCarson, 2020-03-07:
Raspbian Buster 2020-02-13 with all upgrades

Jason2866, 2020-03-08:
Have you tried to install with `sudo` ?

SamuelCarson, 2020-03-09:
yes

From: Jason2866 
Sent: Sunday, March 08, 2020 3:49 PM
To: tasmota/tasmotizer 
Cc: SamuelCarson ; Author 
Subject: Re: [tasmota/tasmotizer] Installation failure Raspbian buster. (#26)

Have you tried to install with sudo ?

Jason2866, 2020-03-09:
Have no (free) RasPi to test. All my RasPis are running headless and are mounted.

jziolkowski, 2020-03-10:
upgrade pip, setuptools and wheel to latest versions and try again

SamuelCarson, 2020-03-30:
upgraded pip, setuptools , and wheel (sudo python3 -m pip install --upgrade pip setuptools wheel) Completed normally. Also upgraded raspbian buster to latest.
Same failure.
ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpnjuqm7ls
         cwd: /tmp/pip-install-q6cbrsa1/PyQt5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 103, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 257, in <module>
        main()
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 240, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 108, in prepare_metadata_for_build_wheel
        config_settings)
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 136, in _get_wheel_metadata_from_wheel
        whl_basename = backend.build_wheel(metadata_directory, config_settings)
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/sipbuild/api.py", line 51, in build_wheel
        project = AbstractProject.bootstrap('pep517')
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/sipbuild/abstract_project.py", line 82, in bootstrap
        project.setup(pyproject, tool, tool_description)
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 387, in setup
        self.apply_user_defaults(tool)
      File "project.py", line 62, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/pyqtbuild/project.py", line 86, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 202, in apply_user_defaults
        self.builder.apply_user_defaults(tool)
      File "/tmp/pip-build-env-kgh9c_rr/overlay/lib/python3.7/site-packages/pyqtbuild/builder.py", line 68, in apply_user_defaults
        "specify a working qmake or add it to PATH")
    sipbuild.pyproject.PyProjectOptionException
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpnjuqm7ls Check the logs for full command output.
pi@raspberrypi:~ $ 

From: Jacek Ziółkowski 
Sent: Tuesday, March 10, 2020 10:00 AM
To: tasmota/tasmotizer 
Cc: SamuelCarson ; Author 
Subject: Re: [tasmota/tasmotizer] Installation failure Raspbian buster. (#26)

upgrade pip, setuptools and wheel to latest versions and try again

jziolkowski, 2020-07-27:
Closing due to lack of activity.

closed 2020-07-27 by jziolkowski

L4RGER, 2020-08-07:
@SamuelCarson did you get this resolved. i have the same issue

L4RGER, 2020-08-08:
pip setuptools and wheel all up to date
`pi@raspberrypi:~ $ sudo pip3 install --upgrade pip setuptools wheel
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in /usr/local/lib/python3.7/dist-packages (20.2.1)
Requirement already up-to-date: setuptools in /usr/local/lib/python3.7/dist-packages (49.2.1)
Requirement already up-to-date: wheel in /usr/local/lib/python3.7/dist-packages (0.34.2)`

still get error 

`pi@raspberrypi:~ $ sudo pip3 install --upgrade pip && sudo pip3 install pyqt5
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in /usr/local/lib/python3.7/dist-packages (20.2.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyqt5
  Using cached PyQt5-5.15.0.tar.gz (3.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpwff8s9nq
         cwd: /tmp/pip-install-uwv2szrl/pyqt5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 126, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
        main()
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 131, in prepare_metadata_for_build_wheel
        config_settings)
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 159, in _get_wheel_metadata_from_wheel
        whl_basename = backend.build_wheel(metadata_directory, config_settings)
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/sipbuild/api.py", line 51, in build_wheel
        project = AbstractProject.bootstrap('pep517')
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/sipbuild/abstract_project.py", line 82, in bootstrap
        project.setup(pyproject, tool, tool_description)
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 477, in setup
        self.apply_user_defaults(tool)
      File "project.py", line 62, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/pyqtbuild/project.py", line 86, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 223, in apply_user_defaults
        self.builder.apply_user_defaults(tool)
      File "/tmp/pip-build-env-2f8h8f6h/overlay/lib/python3.7/site-packages/pyqtbuild/builder.py", line 68, in apply_user_defaults
        "specify a working qmake or add it to PATH")
    sipbuild.pyproject.PyProjectOptionException
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpwff8s9nq Check the logs for full command output.`

SamuelCarson, 2020-08-08:
never resolved!

From: L4RGER 
Sent: Friday, August 07, 2020 7:04 PM
To: tasmota/tasmotizer 
Cc: SamuelCarson ; Mention 
Subject: Re: [tasmota/tasmotizer] Installation failure Raspbian buster. (#26)

@SamuelCarson did you get this resolved. i have the same issue

L4RGER, 2020-08-11:
@SamuelCarson follow the steps outlined in [this](https://stackoverflow.com/questions/55062050/install-pyqt5-on-raspberry-for-python3-6) however i did update to the latest release links of sip and PyQt. takes quite a while, mine is still going 
then after that i'm going to try and run; hopefully with more luck
`pip3 install tasmotizer`

L4RGER, 2020-08-11:
@SamuelCarson as an update, the above steps allowed me to install tasmotizer but not run it. additionally i had to install another package and then it worked fine. what a process but i got there haha.
`sudo apt-get install python3-pyqt5.qtserialport`
hope this helps someone else
this was on a rpi 4b+

mintgroen, 2020-10-09:
> @SamuelCarson as an update, the above steps allowed me to install tasmotizer but not run it. additionally i had to install another package and then it worked fine. what a process but i got there [...]

Was the Qyqt5 missing or outdated in your case?

Updating did not solve my issue:

`pi@raspberrypi:~/tasmotizer $ sudo apt-get install python3-pyqt5.qtserialport
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pyqt5.qtserialport is already the newest version (5.11.3+dfsg-1+rpi1+b3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

`pi@raspberrypi:~/tasmotizer $ tasmotizer.py 
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ValueError: PyCapsule_GetPointer called with incorrect name
`

StoopidWirez, 2020-12-07:
> Was the Qyqt5 missing or outdated in your case?
>
> Updating did not solve my issue:
>
> `pi@raspberrypi:~/tasmotizer $ sudo apt-get install python3-pyqt5.qtserialport Reading package lists... Done [...]

Wondering if you've found a way to run Tasmotizer yet? I'm stuck with this problem right now as well

mintgroen, 2020-12-07:
I've found a workaround, an old windows laptop did the trick...

StoopidWirez, 2020-12-07:
That was my next plan. Was hoping to have it working with my Pi instead but
I appreciate the heads up. Did you have any problems with the windows
version install I should leave ok out for?

mintgroen, 2020-12-07:
Windows installed and works flawlessly, I use the tasmotizer to flash all esp firmwares, tasmota and espeasy.

I would, love to get it working on the rpi, but for some reason none of the found solutions proved successful.

alfredosola, 2020-12-22:
I had the same problem. Raspberry 3B, RaspberryOS updated as of Dec 2020.
Following the solution pointed to by @L4RGER I installed (apt install): `qt5-default pyqt5-dev pyqt5-dev-tools sip-dev`
After this, `sudo pip3 install tasmotizer` ran without errors.

martinrevert, 2021-01-05:
+1 with the same problem on Raspberry Pi 4 and Raspberry OS.

Jason2866, 2021-01-05:
@martinrevert  @alfredosola  It looks like a issue from Raspbery OS. You should open a issue in there Github

dannutu, 2021-06-28:
Just wondering - could this be the explanation? https://github.com/biolab/orange3/issues/5070#issuecomment-869553616

I have to admit that right now I am utterly confused about all these python qt and sip packages and which one does what (so much so that I'd partially echo the last part of https://github.com/biolab/orange3/issues/5070#issuecomment-723450346). 

Installed by the system-wide package manager (apt/dpkg) I have:
```
root@raspberrypi:~# dpkg -l | egrep -i pyqt
ii  pyqt5-dev                            5.11.3+dfsg-1+rpi1                      all          Development files for PyQt5
ii  pyqt5-dev-tools                      5.11.3+dfsg-1+rpi1+b3                   armhf        Development tools for PyQt5
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module
```

Installed by pip I have:
```
root@raspberrypi:~# pip3 list | egrep -i pyqt
PyQt5             5.15.4
PyQt5-sip         12.9.0
```

My gut instinct would be to remove all the 5.11.3 versions from apt/dpkg but I'm really worried this is going to break something else somewhere else. Any (other) ideas?

dannutu, 2021-06-28:
This https://github.com/ros-visualization/rqt_graph/issues/51 and this https://stackoverflow.com/questions/55794101/valueerrorpycapsule-getpointer-called-with-incorrect-name-with-from-pyqt5-qtwe talk about the same issue. There are 2 options:
1. "uninstalling PyQt5 and PyQt5-sip with pip3"
2. "downgrade PyQt5" (via pip install PyQt5==version)

I reckon I could try but https://github.com/tasmota/tasmotizer/blob/release/requirements.txt specifies PyQt5==5.14.2 is required and since the system-wide apt/dpkg pyqt5 versions are all 5.11.3 I'm almost sure it's not going to make tasmotizer work.

jziolkowski, 2021-06-28:
Go ahead with downgrading PyQt's version as you need. Basically any 5.x should be fine as Tasmotizer doesn't use and specific functions of the latest versions

dannutu, 2021-06-28:
Thanks Jacek. I suppose you meant downgrade PyQt using pip3, correct? I.e. not apt/dpkg.
Ok, I just tried to do that using the same version number as the one from the dpkg - 5.11.3 - and it didn't work:
```
root@raspberrypi:~# pip3 list | egrep -i pyqt
PyQt5             5.15.4
PyQt5-sip         12.9.0

root@raspberrypi:~# dpkg -l | egrep -i pyqt
ii  pyqt5-dev                            5.11.3+dfsg-1+rpi1                      all          Development files for PyQt5
ii  pyqt5-dev-tools                      5.11.3+dfsg-1+rpi1+b3                   armhf        Development tools for PyQt5
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module

root@raspberrypi:~# pip3 install PyQt5==5.11.3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement PyQt5==5.11.3 (from versions: 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2, 5.15.3, 5.15.4)
ERROR: No matching distribution found for PyQt5==5.11.3
```

Is there a pip trick to convince it to downgrade to 5.11.3 or should I try one of the listed versions? How should I pick it?

Cheers!

jziolkowski, 2021-06-28:
Altering the requirements.txt should be enough. 
If you can, remove anything PyQt related that was installed via apt.

dannutu, 2021-06-28:
Thanks again. Ok, so I removed the 4 pyqt-related apt/dpkg packages - no issues doing this as they were not required before, I only installed them today for tasmotizer. Same error though:
```
root@raspberrypi:/var/log# dpkg -l | egrep -i pyqt
ii  pyqt5-dev                            5.11.3+dfsg-1+rpi1                      all          Development files for PyQt5
ii  pyqt5-dev-tools                      5.11.3+dfsg-1+rpi1+b3                   armhf        Development tools for PyQt5
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module

root@raspberrypi:/var/log# apt-get remove pyqt5-dev pyqt5-dev-tools python3-pyqt5 python3-pyqt5.qtserialport
The following packages will be REMOVED:
  pyqt5-dev pyqt5-dev-tools python3-pyqt5 python3-pyqt5.qtserialport
0 upgraded, 0 newly installed, 4 to remove and 58 not upgraded.
Removing pyqt5-dev (5.11.3+dfsg-1+rpi1) ...
Removing pyqt5-dev-tools (5.11.3+dfsg-1+rpi1+b3) ...
Removing python3-pyqt5.qtserialport (5.11.3+dfsg-1+rpi1+b3) ...
Removing python3-pyqt5 (5.11.3+dfsg-1+rpi1+b3) ...

root@raspberrypi:/var/log# dpkg -l | egrep -i pyqt

root@raspberrypi:/var/log# tasmotizer.py
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'
root@raspberrypi:/var/log#
root@raspberrypi:/var/log#
```

Should I also remove tasmotizer using pip3 and then try installing it again? Is there anything else I'd also need to clean? I'm not clear at all about all those "wheels" and sip "specifications".

jziolkowski, 2021-06-28:
No no, just install the missing PyQt5.QtSerialPort
It appears it is no longer installed by default and should be added to the requirements

dannutu, 2021-06-28:
Apologies if the questions are trivial, as I mentioned I am not familiar at all with these software packages.

So, I tried installing PyQt5.QtSerialPort using pip3, it wasn't found:
```
root@raspberrypi:~# pip3 list | egrep -i qt
PyQt5             5.15.4
PyQt5-sip         12.9.0

root@raspberrypi:~# pip3 install PyQt5.QtSerialPort
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement PyQt5.QtSerialPort (from versions: none)
ERROR: No matching distribution found for PyQt5.QtSerialPort

root@raspberrypi:~# pip3 install pyqt5.qtserialport
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement pyqt5.qtserialport (from versions: none)
ERROR: No matching distribution found for pyqt5.qtserialport

root@raspberrypi:~# pip3 install python3-pyqt5.qtserialport
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement python3-pyqt5.qtserialport (from versions: none)
ERROR: No matching distribution found for python3-pyqt5.qtserialport
```

On the other hand If I try to reinstall it using apt/dpkg the package manager also wants to install python3-pyqt5 - is that ok? Earlier you mentioned "remove anything PyQt related that was installed via apt"
```
root@raspberrypi:~# apt-cache search PyQt5.QtSerialPort
python-pyqt5.qtserialport - Python 2 bindings for QtSerialPort module
python-pyqt5.qtserialport-dbg - Python 2 bindings for QtSerialPort module (debug extension)
python3-pyqt5.qtserialport - Python 3 bindings for QtSerialPort module
python3-pyqt5.qtserialport-dbg - Python 3 bindings for QtSerialPort module (debug extension)

root@raspberrypi:~# apt-get install python3-pyqt5.qtserialport
The following additional packages will be installed:
  python3-pyqt5
The following NEW packages will be installed:
  python3-pyqt5 python3-pyqt5.qtserialport
0 upgraded, 2 newly installed, 0 to remove and 58 not upgraded.
Do you want to continue? [Y/n] n
Abort.
```

dannutu, 2021-06-28:
I tried it anyway (what did I have to lose?) but it didn't work:

```
root@raspberrypi:~# apt-get install python3-pyqt5.qtserialport
The following additional packages will be installed:
  python3-pyqt5
The following NEW packages will be installed:
  python3-pyqt5 python3-pyqt5.qtserialport
0 upgraded, 2 newly installed, 0 to remove and 58 not upgraded.
Do you want to continue? [Y/n] Y
Selecting previously unselected package python3-pyqt5.
(Reading database ... 190221 files and directories currently installed.)
Preparing to unpack .../python3-pyqt5_5.11.3+dfsg-1+rpi1+b3_armhf.deb ...
Unpacking python3-pyqt5 (5.11.3+dfsg-1+rpi1+b3) ...
Selecting previously unselected package python3-pyqt5.qtserialport.
Preparing to unpack .../python3-pyqt5.qtserialport_5.11.3+dfsg-1+rpi1+b3_armhf.deb ...
Unpacking python3-pyqt5.qtserialport (5.11.3+dfsg-1+rpi1+b3) ...
Setting up python3-pyqt5 (5.11.3+dfsg-1+rpi1+b3) ...
Setting up python3-pyqt5.qtserialport (5.11.3+dfsg-1+rpi1+b3) ...

root@raspberrypi:~# dpkg -l | egrep -i pyqt
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module

root@raspberrypi:~# tasmotizer.py
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ValueError: PyCapsule_GetPointer called with incorrect name
``` 

Now if I want to uninstall python3-pyqt5 it will also uninstall python3-pyqt5.qtserialport, which you said it's needed - kind of a catch 22:
```
root@raspberrypi:~# apt-get remove python3-pyqt5
The following packages will be REMOVED:
  python3-pyqt5 python3-pyqt5.qtserialport
0 upgraded, 0 newly installed, 2 to remove and 58 not upgraded.
Do you want to continue? [Y/n] n
Abort.
```

dannutu, 2021-06-28:
I wonder if this has anything to do with it - both files contain references to Qt:

```
root@raspberrypi:~# find / -iname \*sip\*.so\* -ls | egrep -v libgphoto
find: ‘/run/user/1000/gvfs’: Permission denied
   179592    120 -rw-r--r--   1 root     root       120012 Mar 31  2019 /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
   179790    628 -rwxr-xr-x   1 root     staff      642576 Jun 28 05:33 /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so

root@raspberrypi:~# strings /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so | egrep -ci qt
14
root@raspberrypi:~# strings /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so | egrep -ci qt
25

root@raspberrypi:~# strings /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so | egrep -i qt
sipQtSupport
sipQtSupport->qt_find_sipslot
pyqt_kw_handler
/build/sip4-hgGQdY/sip4-4.19.14+dfsg/siplib/qtlib.c
sipQtSupport->qt_create_universal_slot
sipQtSupport
sipQtSupport->qt_emit_signal
sipQtSupport->qt_same_name
sipQtSupport->qt_find_slot
sipQtSupport->qt_connect
sipQtSupport->qt_connect_py_signal
sipQtSupport->qt_disconnect
sipQtSupport->qt_destroy_universal_slot
sipQtSupport->qt_disconnect_py_signal

root@raspberrypi:~# strings /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so | egrep -i qt
sipQtSupport
pyqt_kw_handler
PyQt5.sip._C_API
PyQt5.sip
qtlib.c
sipQtSupport
_sipQtAPI
qt_find_sipslot
qt_connect
qt_emit_signal
qt_create_universal_signal
qt_disconnect
qt_find_slot
qt_disconnect_py_signal
qt_connect_py_signal
qt_qobject
qt_same_name
qt_find_universal_signal
qt_destroy_universal_slot
qt_create_universal_slot
/tmp/pip-wheel-hy466tvq/pyqt5-sip_b741eac94f9b440aa1cacd1db282fed6
em_qt_api
qtlib.c
qtlib.c
sipQtSupport

```

dannutu, 2021-06-28:
> Altering the requirements.txt should be enough.

Could you please let me know how and when can I do this? "pip3 install tasmotizer" downloads and processes everything from the server (including requirements.txt) so I don't have any opportunity to intervene. Should I then try "Option 3: Clone the repo, pip3 install -r requirements.txt followed by python3 tasmotizer.py and flash away!"?

jziolkowski, 2021-06-28:
yes, try method 3 and modify the file

also did you try `apt install python3-pyqt5.qtserialport` ? I can't find a way to install that module via pip, that's weird

dannutu, 2021-06-28:
> yes, try method 3 and modify the file

Ok, I could try but what exactly should I include in the requirements.txt file? 
```
  pyserial>=3.5
  PyQt5>=5.0.0
```
?

I'm asking because I have serious doubts it will work - ``pip3 install -v PyQt5==5.11.3`` showed that 112 versions of PyQt from 5.6 to 5.15.4 for different architectures (e.g. win32, win_amd64, macosx etc.) were checked and they all failed the check with ``Skipping link: none of the wheel's tags ([tags omitted for clarity]) are compatible (run pip debug --verbose to show compatible tags)``

``pip3 debug --verbose`` listed 248 compatible tags, of which 238 end in ``armv7l`` and 10 end in:
```
py3-none-any
py37-none-any
py36-none-any
py35-none-any
py34-none-any
py33-none-any
py32-none-any
py31-none-any
py30-none-any
cp37-none-any
```

Also, not sure how to get the required pyqt5.qtserialport installed without also getting python3-pyqt5 installed (which will end up being version 5.11.3+dfsg-1+rpi1+b3). Which nicely brings me to your second point:

> also did you try `apt install python3-pyqt5.qtserialport` ? I can't find a way to install that module via pip, that's weird

Yes, I tried earlier, see https://github.com/tasmota/tasmotizer/issues/26#issuecomment-869868915 above. It resulted in the same error - "PyCapsule_GetPointer called with incorrect name"

I'm out of ideas to try :(

dannutu, 2021-06-28:
For completeness, here are the relevant fragments from ``pip3 install -v PyQt5==**5.11.3**``:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2 location(s) to search for versions of pyqt5:
* https://pypi.org/simple/pyqt5/
* https://www.piwheels.org/simple/pyqt5/
Fetching project page and analyzing links: https://pypi.org/simple/pyqt5/
Getting page https://pypi.org/simple/pyqt5/
Found index url https://pypi.org/simple
Looking up "https://pypi.org/simple/pyqt5/" in the cache
Request header has "max_age" as 0, cache bypassed
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/pyqt5/ HTTP/1.1" 304 0```

```
  Skipping link: none of the wheel's tags (cp35-abi3-manylinux1_x86_64, cp36-abi3-manylinux1_x86_64, cp37-abi3-manylinux1_x86_64, cp38-abi3-manylinux1_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/d4/bf/d884da8e2f7096d201c891d515eb6813a8e85df5eb6f5e12e867bf1d831c/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl#sha256=ac1eb5a114b6e7788e8be378be41c5e54b17d5158994504e85e43b5fca006a39 (from https://pypi.org/simple/pyqt5/)
  Skipping link: none of the wheel's tags (cp35-none-win32, cp36-none-win32, cp37-none-win32, cp38-none-win32) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/26/78/c215008e70982944272b6c329a76cc11259e7378a3e23418fbe0bd48de96/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-none-win32.whl#sha256=e85936bae1581bcb908847d2038e5b34237a5e6acc03130099a78930770e7ead (from https://pypi.org/simple/pyqt5/)
  Skipping link: none of the wheel's tags (cp35-none-win_amd64, cp36-none-win_amd64, cp37-none-win_amd64, cp38-none-win_amd64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/a7/2d/d2c989006c86ae98ed230c28c3e0dd7fa0374e723afc107d12268159ceb7/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-none-win_amd64.whl#sha256=d2309296a5a79d0a1c0e6c387c30f0398b65523a6dcc8a19cc172e46b949e00d (from https://pypi.org/simple/pyqt5/)
  Skipping link: none of the wheel's tags (cp35-abi3-macosx_10_6_intel, cp36-abi3-macosx_10_6_intel, cp37-abi3-macosx_10_6_intel, cp38-abi3-macosx_10_6_intel) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/31/22/e79a35bab2221b7bdbb3cdadb25bc9b492080b7529eec5fcbfd3f2d57606/PyQt5-5.11.3-5.11.2_a-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl#sha256=517e4339135c4874b799af0d484bc2e8c27b54850113a68eec40a0b56534f450 (from https://pypi.org/simple/pyqt5/)
ERROR: Could not find a version that satisfies the requirement PyQt5==5.11.3 (from versions: 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2, 5.15.3, 5.15.4)
ERROR: No matching distribution found for PyQt5==5.11.3
pip._vendor.resolvelib.resolvers.RequirementsConflicted: Requirements conflict: SpecifierRequirement('PyQt5==5.11.3')
pip._vendor.resolvelib.resolvers.ResolutionImpossible: [RequirementInformation(requirement=SpecifierRequirement('PyQt5==5.11.3'), parent=None)]
pip._internal.exceptions.DistributionNotFound: No matching distribution found for PyQt5==5.11.3```

The 112 version checked:
```
PyQt5-5.6-cp35-cp35m-macosx_10_6_intel.whl
PyQt5-5.6-cp35-cp35m-manylinux1_x86_64.whl
PyQt5-5.6-cp35-none-win32.whl
PyQt5-5.6-cp35-none-win_amd64.whl
PyQt5-5.7-cp35-cp35m-macosx_10_6_intel.whl
PyQt5-5.7-cp35-cp35m-manylinux1_x86_64.whl
PyQt5-5.7-cp35-none-win32.whl
PyQt5-5.7-cp35-none-win_amd64.whl
PyQt5-5.7.1-5.7.1-cp34.cp35.cp36-abi3-macosx_10_6_intel.whl
PyQt5-5.7.1-5.7.1-cp34.cp35.cp36-abi3-manylinux1_x86_64.whl
PyQt5-5.7.1-5.7.1-cp34.cp35.cp36-none-win_amd64.whl
PyQt5-5.7.1-5.7.1-cp34.cp35.cp36.cp37-none-win32.whl
PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win32.whl
PyQt5-5.8-5.8.0-cp34.cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-none-win32.whl
PyQt5-5.8.1.1-5.8.0-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-none-win32.whl
PyQt5-5.8.2-5.8.0-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.9-5.9.1-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.9-5.9.1-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.9-5.9.1-cp35.cp36.cp37-none-win32.whl
PyQt5-5.9-5.9.1-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win32.whl
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-none-win32.whl
PyQt5-5.9.2-5.9.3-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.10-5.10.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl
PyQt5-5.10-5.10.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl
PyQt5-5.10-5.10.0-cp35.cp36.cp37-none-win32.whl
PyQt5-5.10-5.10.0-cp35.cp36.cp37-none-win_amd64.whl
PyQt5-5.10.1-5.10.1-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl
PyQt5-5.10.1-5.10.1-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
PyQt5-5.10.1-5.10.1-cp35.cp36.cp37.cp38-none-win32.whl
[... cut ...]
PyQt5-5.15.2-5.15.2-cp35.cp36.cp37.cp38.cp39-none-win_amd64.whl
PyQt5-5.15.3-cp36.cp37.cp38.cp39-abi3-macosx_10_13_intel.whl
PyQt5-5.15.3-cp36.cp37.cp38.cp39-abi3-manylinux2014_x86_64.whl
PyQt5-5.15.3-cp36.cp37.cp38.cp39-none-win32.whl
PyQt5-5.15.3-cp36.cp37.cp38.cp39-none-win_amd64.whl
PyQt5-5.15.4-cp36.cp37.cp38.cp39-abi3-macosx_10_13_intel.whl
PyQt5-5.15.4-cp36.cp37.cp38.cp39-abi3-manylinux2014_x86_64.whl
PyQt5-5.15.4-cp36.cp37.cp38.cp39-none-win32.whl
PyQt5-5.15.4-cp36.cp37.cp38.cp39-none-win_amd64.whl```

jziolkowski, 2021-06-29:
> ```
>   PyQt5>=5.0.0
> ```

Try the exact same version that you used earlier using ==
When you have installed 5.11.x manually and then install tasmotizer via pip, it will upgrade it. Pinning the version in the requirements using == should at least ensure you're not getting the newer version.
Start with 5.10 maybe? IDK

dannutu, 2021-06-29:
Ok, do I need to first clean anything up (e.g. uninstall any packages or delete any files) before attempting this? Otherwise I think some files are going to retain info from the already installed versions, e.g. the second one of these, which was built yesterday, it contains references to PyQt5 and sip and it doesn't belong to any currently installed package:
```
root@raspberrypi:~/tmp# ls -ald /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
-rw-r--r-- 1 root root  120012 Mar 31  2019 /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
-rwxr-xr-x 1 root staff 642576 Jun 28 05:33 /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so

root@raspberrypi:~/tmp# dpkg -S /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
python3-sip: /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
root@raspberrypi:~/tmp# dpkg -S /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
dpkg-query: no path found matching pattern /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so

root@raspberrypi:~/tmp# dpkg-query -S /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
python3-sip: /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
root@raspberrypi:~/tmp# dpkg-query -S /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
dpkg-query: no path found matching pattern /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so

root@raspberrypi:~/tmp# dpkg-query --search /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
python3-sip: /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
root@raspberrypi:~/tmp# dpkg-query --search /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
dpkg-query: no path found matching pattern /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so

root@raspberrypi:~/tmp# apt-file search /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
python3-sip: /usr/lib/python3/dist-packages/sip.cpython-37m-arm-linux-gnueabihf.so
root@raspberrypi:~/tmp# apt-file search /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
```

jziolkowski, 2021-06-29:
All stuff that is no longer needed can be removed by `apt autoremove`

dannutu, 2021-07-11:
> All stuff that is no longer needed can be removed by `apt autoremove`

Ok, I found a little bit of time to look at this again.

I removed all pyqt packages from the system:

```
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i pyqt
PyQt5             5.15.4
PyQt5-sip         12.9.0
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 uninstall PyQt5-sip
Found existing installation: PyQt5-sip 12.9.0
Uninstalling PyQt5-sip-12.9.0:
  Would remove:
    /usr/local/lib/python3.7/dist-packages/PyQt5/sip.cpython-37m-arm-linux-gnueabihf.so
    /usr/local/lib/python3.7/dist-packages/PyQt5_sip-12.9.0.dist-info/*
Proceed (y/n)? y
  Successfully uninstalled PyQt5-sip-12.9.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i pyqt
PyQt5             5.15.4
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 uninstall PyQt5
Found existing installation: PyQt5 5.15.4
Uninstalling PyQt5-5.15.4:
  Would remove:
    /usr/local/bin/pylupdate5
    /usr/local/bin/pyrcc5
    /usr/local/bin/pyuic5
    /usr/local/lib/python3.7/dist-packages/PyQt5-5.15.4.dist-info/*
    /usr/local/lib/python3.7/dist-packages/PyQt5/*
Proceed (y/n)? y
  Successfully uninstalled PyQt5-5.15.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i pyqt
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i qt
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#

root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# dpkg -l | egrep -i pyqt
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# apt remove python3-pyqt5.qtserialport python3-pyqt5
[...]
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# apt autoremove
[...]

root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# dpkg -l | egrep -i pyqt
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i qt
```

I then tried manually installing PyQt5 5.11.3 via pip3 - it couldn't find it:

```
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 install -v PyQt5==5.11.3
[...]
ERROR: Could not find a version that satisfies the requirement PyQt5==5.11.3 (from versions: 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2, 5.15.3, 5.15.4)
ERROR: No matching distribution found for PyQt5==5.11.3
```

I listed all the available pyqt versions:
```
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 install PyQt5==
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement PyQt5== (from versions: 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2, 5.15.3, 5.15.4)
ERROR: No matching distribution found for PyQt5==
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
[... 62 lines cut ...]
ii  python3-sip-dev                      4.19.14+dfsg-2                          armhf        Python 3/C++ bindings generator development files
ii  sip-dev                              4.19.14+dfsg-2                          armhf        Python/C++ bindings generator code generator application
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#

root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# apt install python3-sip python3-sip-dev sip-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-sip is already the newest version (4.19.14+dfsg-2).
python3-sip set to manually installed.
python3-sip-dev is already the newest version (4.19.14+dfsg-2).
python3-sip-dev set to manually installed.
sip-dev is already the newest version (4.19.14+dfsg-2).
0 upgraded, 0 newly installed, 0 to remove and 63 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
```

Which I think was the problem all along, since the previously installed pyqt (5.15.4) most likely also needed at least 'sip >=5.0.1'.

Unfortunately I don't have time to keep digging to try to get this to work. I already used esptool.py 2 weeks ago, I will keep using it - it just works.

dannutu, 2021-07-11:
Just before letting this go for good - I tried installing the latest PyQt version (5.15.4) via pip3 in the venv and changing requirements.txt to use it:
```
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# venv01/bin/pip3 install PyQt5==5.15.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting PyQt5==5.15.4
  Using cached PyQt5-5.15.4-cp36.cp37.cp38.cp39-abi3-manylinux2014_armv7l.whl
Collecting PyQt5-sip<13,>=12.8
  Using cached https://www.piwheels.org/simple/pyqt5-sip/PyQt5_sip-12.9.0-cp37-cp37m-linux_armv7l.whl (274 kB)
Installing collected packages: PyQt5-sip, PyQt5
Successfully installed PyQt5-5.15.4 PyQt5-sip-12.9.0
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# dpkg -l | egrep -i pyqt
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# ./venv01/bin/pip3 list | egrep -i pyqt
PyQt5         5.15.4
PyQt5-sip     12.9.0
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# vi requirements.txt
[...]
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# cat requirements.txt
pyserial>=3.5
PyQt5>=5.14.0
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# ./venv01/bin/pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyserial>=3.5
  Using cached https://www.piwheels.org/simple/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
Requirement already satisfied: PyQt5>=5.14.0 in ./venv01/lib/python3.7/site-packages (from -r requirements.txt (line 2)) (5.15.4)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in ./venv01/lib/python3.7/site-packages (from PyQt5>=5.14.0->-r requirements.txt (line 2)) (12.9.0)
Installing collected packages: pyserial
Successfully installed pyserial-3.5
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# ./venv01/bin/python3 tasmotizer.py
Traceback (most recent call last):
  File "tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#
```

PyQt5-5.15.4 installed ok but tasmotizer.py throws an error because it cannot find the module PyQt5.QtSerialPort

But is PyQt5.QtSerialPort actually needed?? See https://wiki.qt.io/Qt_Serial_Port:
```
NOTE: As of Qt 5.1, Qt Serial Port is now officially part of Qt. You no longer need to build Qt Serial Port manually; simply install Qt 5.1 or later and go to section Usage https://wiki.qt.io/Qt_Serial_Port#Usage
```

dannutu, 2021-07-11:
Wow, I just got it working, mostly by chance! :)

I used "Option 3: Clone the repo". It still doesn't work in the new virtual environment but it does work in the "main" system and that's ok with me, at least for now.

For anybody else trying to get it to work under Raspbian 10 (buster) these are the relevant installed packages.
**Note how there is no pyqt, serialport or sip module installed by pip3 (at least not on the "main" system, there are only 2 modules installed in the virtual environment, which is not used anyway)**
```
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# dpkg -l | grep -i 'pyqt|serialport|sip'
ii  libqt5serialport5:armhf              5.11.3-2                                armhf        Qt 5 serial port support
ii  libserialport0:armhf                 0.1.1-3                                 armhf        Crossplatform serial port handling library - shared library
ii  python3-pyqt5                        5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for Qt5
ii  python3-pyqt5.qtserialport           5.11.3+dfsg-1+rpi1+b3                   armhf        Python 3 bindings for QtSerialPort module
ii  python3-sip                          4.19.14+dfsg-2                          armhf        Python 3/C++ bindings generator runtime library
ii  python3-sip-dev                      4.19.14+dfsg-2                          armhf        Python 3/C++ bindings generator development files
ii  sip-dev                              4.19.14+dfsg-2                          armhf        Python/C++ bindings generator code generator application

root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# pip3 list | egrep -i --color 'pyqt|serialport|sip'
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#

root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer# ./venv01/bin/pip3 list | egrep -i --color 'pyqt|serialport|sip'
PyQt5         5.15.4
PyQt5-sip     12.9.0
root@raspberrypi:/home/pi/Downloads/Tasmota/tasmotizer#

```

I also tried to figure out the entire sequence of apt commands, this is it:
```
2021-06-28:
apt-get install qt5-default pyqt5-dev pyqt5-dev-tools sip-dev python3-pyqt5.qtserialport
apt-get remove pyqt5-dev pyqt5-dev-tools python3-pyqt5 python3-pyqt5.qtserialport
apt-get install python3-pyqt5.qtserialport

2021-07-11:
apt remove python3-pyqt5.qtserialport python3-pyqt5
apt autoremove
apt install python3-pyqt5.qtserialport
apt install python3-serial python3-pyqt5 --reinstall
```

Voronenko, 2021-07-27:
@dannutu  thank you very much for the hint.

I do confirm issue with Debian 10 on a regular secondary notebook I am using for my wired experiments :)

Following 

```
2021-06-28:
apt-get install qt5-default pyqt5-dev pyqt5-dev-tools sip-dev python3-pyqt5.qtserialport
apt-get remove pyqt5-dev pyqt5-dev-tools python3-pyqt5 python3-pyqt5.qtserialport
apt-get install python3-pyqt5.qtserialport

2021-07-11:
apt remove python3-pyqt5.qtserialport python3-pyqt5
apt autoremove
apt install python3-pyqt5.qtserialport
apt install python3-serial python3-pyqt5 --reinstall
```

I got into moment, when it at least tries to compile QT.

with pip install --verbose -r requirements.txt over tasmota repo

Voronenko, 2021-07-27:
@SamuelCarson , @jziolkowski    I would say original issue exists, and not only on a raspberry, but also on a regular notebooks (although quite older)

Jason2866, 2021-07-27:
> @SamuelCarson , @jziolkowski I would say original issue exists, and not only on a raspberry, but also on a regular notebooks (although quite older)

Your statement is unspecific. Tasmotizer does work on many Linux variants.
Which Linux version is used which is not working?

Voronenko, 2021-07-28:
@Jason2866   

I have mentioned "original issue" meaning ticket opened by  @SamuelCarson , i.e.  https://github.com/tasmota/tasmotizer/issues/26

So I do confirm identical installation issue on a Debian Buster 10, and not only on ARM hardware but regular intel one.

Acceptance criteria: following pre-steps mentioned in  https://github.com/tasmota/tasmotizer "Installation and how to run"

```
pip3 install tasmotizer
```

should complete without issues. On Debian 10 codename "buster" it is not (original ticket for Raspbian) and I do confirm that for clean Debian 10 - it is also does not pass.

Although workaround by @dannutu  works for scenario install from repo. So perhaps workaround is a missing installation pre-steps for that problem.