tasmota/tasmotizer #27: Error during execution after clone install

SamuelCarson, 2020-03-07:
pi@raspberrypi:~/Downloads/tasmotizer $ python3 tasmotizer.py
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: palette support is disabled
qt5ct: custom style sheet is disabled

SamuelCarson, 2020-03-07:
Flash works but in "send config"  no fields are writable
Raspian Buster 2020-02-13 with all updates

Jason2866, 2020-03-08:
Have you tried to start with `sudo` ?
QT5 installed?

SamuelCarson, 2020-03-09:
Yes to all

From: Jason2866 
Sent: Sunday, March 08, 2020 3:50 PM
To: tasmota/tasmotizer 
Cc: SamuelCarson ; Author 
Subject: Re: [tasmota/tasmotizer] Error during execution after clone install (#27)

Have you tried to start with sudo ?
QT5 installed?

jziolkowski, 2020-03-10:
You have to tick the small checkboxes next to group name, it enables them for editing and selects them to be sent to device.

Jason2866, 2020-09-01:
See https://github.com/tasmota/tasmotizer/issues/26#issuecomment-672277551

closed 2020-09-01 by Jason2866