tasmota/tasmotizer #29: Send config options always grayed out?

jcwren, 2020-03-27:
Using v1.1, after flashing a device (using the 'Release 8.2.0' radio button), if I go to 'Send config', the options are grayed out. How does one send the configuration?

IngmarG, 2020-03-29:
Same question

jziolkowski, 2020-03-31:
You need to click the small checkbox next to the group name

jcwren, 2020-03-31:
I have to say that a black check box on a very dark background that's right next to the a windows border box is extremely NOT obvious. Maybe they should all be enabled by default, and let the user disable the ones not needed themselves. At least that way it would be obvious that a checkbox is there.

jziolkowski, 2020-03-31:
Yes, that's why in dev version it's been fixed. You can try it but there might be some other issues.

jziolkowski, 2020-03-31:
It will be released soonish

jmuellers, 2021-05-01:
It took me long to track this down and find the checkboxes, they are basically invisible due to the color scheme (I supposed it is different from screen to screen). 

I guess the "soonish" more than one year ago did not happen yet? ;-)

jziolkowski, 2021-05-01:
Unfortunately yes, I'm bogged down with other stuff :(

jmuellers, 2021-05-01:
> Unfortunately yes, I'm bogged down with other stuff :(

Sorry to hear! In any case, your Tasmotizer is a great tool that saved me a lot of work, thanks for all your effort :-)