tasmota/tasmotizer #30: Unable to send config file

JayJay2105, 2020-04-01:
Hi,
Flashed a Sonoff T2uk3ch using tasmotizer, that all works fine, powered off, disconnected gpio from gnd, powered on, try to send config of t1 3ch as per other users, reports sent ok, power off and on, not connecting to wifi, no ip address showing up in router, check in with termite, ipadress1 - reports ipUnSet. Wifi light ono Sonoff just flashes.

jziolkowski, 2020-04-13:
Yes, I've noticed it recently as well. Check the fix in the development version (it will require manual installation of the required packages)

silentcossack, 2020-04-14:
I tried the latest git develop version and I still can't send config. Flashing worked fine, but whenever i try to send config i get this "bool QSerialPort::flush(): device not open". I'm on Mac, using the python3 version.

jziolkowski, 2020-09-21:
Please test again using release 1.2, and develop-1.3