tasmota/tasmotizer #32: Exe file - differences between tasmotiozer.exe and tasmotizer-x86.exe file

bovirus, 2020-04-01:
@jziolkowski

What are the differences between tasmotizer.exe and tasmotizer-x86.exe?

Could include this info in readme?

Thanks.

Jason2866, 2020-04-04:
X86 and x64 version

closed 2020-04-04 by Jason2866

bovirus, 2020-04-04:
@jziolkowski 

tasmotizer.exe = x64 version
tasmotizer_x86.xe = x86 version

Then why don't rename 

**_tasmotizer.exe_** as **_tastnotizer_x64.exe_** 

to highlight these difference.

Please add this info about exe difference in readme.

Thanks.

Jason2866, 2020-04-04:
Because x64 is standard these days

bovirus, 2020-04-04:
@Jason2866 

I will avoid misunderstanding (not all user can understand that x64 is standard exe) and I will rename the exe as _x64.exe.

jziolkowski, 2020-10-01:
For some users their standard is X86, for others it's x64. I'm from the latter.
The version will be named as they are now. I've updated the readme.

Case closed.

ramanraja, 2020-10-02:
OK, thank you.
(And thanks for this wonderful software. We all love it!)
Warm regards,
Rajaraman