tasmota/tasmotizer #33: Pass Phrases with Spaces

dennyfmn, 2020-04-08:
It seems that the Send Config does not work if the Password for the network SSID has spaces in it. It tells me that it sent 246 characters, but nothing happens.

If I send it another SSID and password pair that does not have spaces in the Password, it works.

I tried single and double quoting the Password, but that did not help. It took the quote marks as part of the Password.

I am running tasmotizer-1.1a.exe on Win10.

Thanks!

jziolkowski, 2020-04-13:
if your are able to, please test the developmen branch, it's just been updated.
That will require manual installation of required packages if you're up to it.

dennyfmn, 2020-04-14:
Hi Jacek,

 

I tested the development version on Win10. I used the method 3, manually install the required packages and then execute tasmotizer.py.

 

Here is what I found:

 

I checked to box for Backup original firmware and Erase before flashing.

I selected Release Tasmota-sensors.bin

I put the ESP-01 into program mode and clicked Tasmotize!

It made a backup .bin file but got stuck when trying to proceed with the erase.

I had to restart Tasmotizer, uncheck Backup original firmware and then clicked Tasmotize! This time it erased and flashed the new image.

I followed the instruction to power the adapter down/up.

I clicked on Send config

(BTW, I like the new Send config options!)

After filling in all my choices I clicked Save. It reported that it sent xxx bytes, but when I checked the serial output I found that the ESP-01 had gone into Access Point mode on 192.168.4.1, not on my WiFi network as I had selected.

This time I power cycled the ESP-01 and gave it a reset.

When I went to the Send config page, I saw that it had not remembered some of my selections correctly:

  The Hostname was displayed as “2”, (without the quotes)

   The Template was displayed as “2”, (without the quotes)

Fixed up my selections and clicked Save

It reported sending xxx bytes and this time it did show up on my WiFi network!

Great! My WiFi passphrase does contain spaces, so that works now!

When I brought up the web interface on the ESP-01, I saw the following:

  The template that I had set up was in Configure | Other but it was not Activated. I clicked the Activate box and restated it.

   SetOption65 was set to ON as I had requested

   I had to set a Configuration | Friendly Name. It would be nice if the Send config page had an option to set this.

 

I hope this helps! This is a great tool, thank you for all your hard work!

 

Denny

dennyfmn, 2020-04-14:
I see that there is a Tazmotizer channel on the Tasmota Discord server that you have been posting to. It says that I do not have permission to send messages to this channel. I'd be happy to communicate via Discord if I had permission. Thanks!

jziolkowski, 2020-04-14:
Thanks a lot for this test case, I'll reproduce these steps. I admit, most of my testing is on self-resetting devices like nodemcu or d1 mini, so it's time to solder up some bare ESP modules and test more.

Also thanks for all the kind words. The tasmotizer channel on discord is only for github bot feeds, but you can reach me on the general channel as @Faziaty#6740. I'm pretty much always there, even when invisible

dennyfmn, 2020-04-14:
Great! Thanks! Let me know when there is another version to test…

jziolkowski, 2020-09-21:
The new version has been released today for all platforms, including fixes for your issue. Please test.

closed 2020-09-21 by jziolkowski