tasmota/tasmotizer #34: Feature Request - Option to Save Password/Passphrase

dennyfmn, 2020-04-10:
It would be nice to have the option to save the Password/Passphrase to the .cfg file. This would be very useful when setting up a batch of units on the same network.

jziolkowski, 2020-04-13:
I'll add it in a secure way

richms, 2020-07-05:
Is there a windows exe with this feature? Im getting over pasting it back in all the time.

Kunzol, 2020-07-31:
I have no Windows.

jziolkowski, 2020-09-21:
Check the latest additions in develop-1.3 branch. New send config dialog stores the passwords in the system keyring.

Kunzol, 2020-10-02:
Just tried it and it works as expected. Thanks !!

Of course a gpg agent needs to be running, to store the password in its keyring.

jziolkowski, 2020-10-02:
That depends on the OS, but yes. If the keyring is to be secure it needs some encryption. But you'd probably have it running by default anyway.