tasmota/tasmotizer #39: 1.1a GUI Quits before flashing

richard-scott, 2020-05-01:
I have used the 1.1a as soon as I click on the "release" or "development" selectors in the middle the app quits. I want to be able to select from the dropdown the release to install but the app just quits.

richard-scott, 2020-05-04:
Odd, it's working today.... seems to be network related.

To test that theory, I've just disconnected from the Internet, loaded the app and it quit when trying to select a firmware release to install.

richard-scott, 2020-05-05:
I get this error in the console from the latest build form the develop branch:

[image]

The only way I've been able to stop it quitting is to do the following:

[image]

golddragon007, 2020-05-20:
It seems for me from the error like the decoder doesn't get JSON in time. With another project I had a problem because they set the timeout for 2 seconds and I've needed appr. 4... maybe similar here too, however for me it's perfectly working the selectors.

richard-scott, 2020-05-20:
@golddragon007 yes, I agree the timeout may be too low. To test you could disable Internet access on your computer by switching off WiFi, or disconnecting the network cable. The lack of a JSON response from the Internet isn't handled gracefully.

golddragon007, 2020-05-20:
As I've dug into this now a bit, I just realized, I don't know python well :D. Well, I see there's a lot of places where I can improve this (after now I've flashed 15 devices in a row), just the knowledge missing :(. Besides that, I've fount the location I can't put there anything because the library doesn't support such stuff (like timeout parameter), you need your implementation for that. Soooo... I'm dead. What I've fount it waits around 2 seconds and then it will exit (at least the comment says that).

richard-scott, 2020-05-21:
No worries @golddragon007 I'm going to have a look over the weekend. I know Python a bit, so may be able to fix it.

andykams, 2020-06-21:
I think I am having a similar issue too . 
When I select backup firmware and flash the device I get the progress bar stuck on writing done. 
[image]

Then errors out:
[image]

If I do not check "Back up original firmware" it completes flashing fine.

golddragon007, 2020-06-22:
Ahh, you get an error message? After how much time? Because I think I've waited for half a min and nothing happened. And every time, when it's in this state and I disconnecting the device, it's crashing the application.

mrekin, 2020-07-02:
So the same thing (GUI close) happens when com port is busy by other app\terminal.

andykams, 2020-07-02:
It was about 2-3 mins after I get an error. That was 2 out of 5 times and rest I did not get a error.

Mark-Muc, 2020-07-11:
same issue here by selecting "development", waiting cursor show up for 2 sec and then it "quits" itself
"release" 8.3.1 works as expected

Morphy99, 2020-08-14:
> same issue here by selecting "development", waiting cursor show up for 2 sec and then it "quits" itself
> "release" 8.3.1 works as expected

Have you got a link for 8.3.1 please? I'm trying to use this to backup an ESP8266 and without using ESPTOOL the only way on windowz

richard-scott, 2020-08-14:
They are on Github:

https://github.com/arendst/Tasmota/releases/tag/v8.3.1

Morphy99, 2020-08-14:
> They are on Github:
> 
> https://github.com/arendst/Tasmota/releases/tag/v8.3.1

Sorry, I thought it was an earlier version of Tasmotizer! I'm not backing up tasmota, using it to backup WLED and ESPHome. Guess I'll have to boot into Linux live and use esptool command line to backup the ESP for now until this is fixed.

theetron, 2020-08-27:
Sam issue here, after selecting release or develop, application crashes. What can be done to fix this?

PreferHardware, 2020-08-27:
Same here: open Tasmotizer on Windows and Linux (Mint) and it crashes as soon as Release or Development is selected. On Windows it just closes and on Linux I get:

Traceback (most recent call last):
  File "tasmotizer.py", line 557, in setBinMode
    self.processReleaseInfo()
  File "tasmotizer.py", line 577, in processReleaseInfo
    reply = json.loads(str(self.release_data, 'utf8'))
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Mark-Muc, 2020-08-28:
same by me .. now it "chrashes" also selecting release .. I guess the URL's to release and Developemd  has changed ..

PreferHardware, 2020-08-28:
In https://tasmota.github.io/docs/ and elsewhere the download location for Tasmota binaries is linked to http://thehackbox.org/tasmota/release/. This is no longer accessible. On reflection, I wonder if this is what is causing Tasmotize to fail.

As a workaround, I used the NodeMCU Py-Flasher from https://github.com/marcelstoer/nodemcu-pyflasher/releases and manually downloaded the binary I needed from https://github.com/arendst/Tasmota/releases/tag/v8.4.0. This worked well for me.

Jason2866, 2020-08-29:
Can you try latest release version?

PreferHardware, 2020-08-29:
Hi Jason, I've just tried 1.1c and it works as designed although I've not checked the version it actually downloads, but since you've changed the source to the OTA repo I'm assuming it is the latest release version. Well done, so quick, thank you. Resolved.

discip, 2020-08-29:
@Jason2866
Good evening sir,
**thank you** for the almost *immediate* response!
But every single release beginning with ***`tasmotizer-1.1a.exe`*** is behaving the same when either 'Release' or 'Development' is selected: It just closes.
The only working one is the ***`tasmotizer-1.0.exe`***.

I am on latest **Win10** *(64bit, if that matters)*.

regards

PreferHardware, 2020-08-29:
1.1c has just worked for me.

discip, 2020-08-29:
@PreferHardware 
Good evening sir,
may I please get to know, what you are running?

PreferHardware, 2020-08-29:
W10-64. Just checked again: I get the release/development drop-down lists, where 1.1a quit.

discip, 2020-08-29:
Strange!
Have you plugged in any any 'flash-able' device?

PreferHardware, 2020-08-29:
No, the first time. Yes, the second. It works either way.

discip, 2020-08-29:
That gives me a hard time, because there seems to be no way for me to get it to work.
I re-downloaded it at least once more and even tried your link, but both ways without any luck.
The ***1.0*** however works as expected.

Jason2866, 2020-08-29:
@discip Have you checked Backup or Erase? Try with uncheck both.

theetron, 2020-08-30:
> Same issue here, after selecting release or develop, application crashes. What can be done to fix this?

Version 1.1a is working now, and the 1.1c is also working.

discip, 2020-08-31:
@Jason2866 
I have tried every possible combination with and without a device being plugged in.
It just keeps closing. 🤷‍♂️😵

By the way: Backup saves only 1MB. I have used a Shelly2.5 and its backup should be 2MB.

thanks
regards

Jason2866, 2020-08-31:
The new version (no release date yet) will have a option to do different backup sizes

discip, 2020-09-01:
@Jason2866
*Sorry for off topic content.*
Will it detect the flash size by itself, like "[NodeMCU-PyFlasher-4.0](https://github.com/marcelstoer/nodemcu-pyflasher/releases)" does?
If you check **`Erase before flashing`** does it erase only 1MB or according to the actual flash size of the attached device?

discip, 2020-09-01:
@Jason2866
Now the lastest Version ***1.1c*** works as intended.
All I did was executing different flashing software like:

1. **ESP.Easy.Flasher.exe** & its updater ***Update.exe***
2. **NodeMCU-PyFlasher-4.0-x64.exe**
3. **Tasmota-PyFlasher-1.0.exe**

Just in case someone stumbles across the issue I had as well.
Maybe somebody is able to figure out, why it suddenly started to work, after I did what is mentioned above. Maybe the ***Updater.exe*** did the trick. 🤷‍♂️

Jason2866, 2020-09-01:
Flash erase does erase the whold flash. It does not matter if it is 1M 2M 4M or 16M

closed 2020-09-01 by Jason2866

discip, 2020-09-01:
@Jason2866
## Thank you for you consistency in replying. Please keep it this way. 👍😀
I hope yo did not get me wrong, about me, being some kind of taciturn in terms of saying THANK YOU for your effort, while mentioning the issues I had.
It was due to the fact that there are **so many issues opened** and it seemed the **developer dropped this project**, than, all of a sudden, YOU popped out of nowhere (I did not know until recently, that you were involved in developing this besides @jziolkowski.), so I thought HE (meaning you) might be the one, being able to solve all the grievances, and I jumped just right at you. 😔
# 👍THANK YOU 👍

Jason2866, 2020-09-01:
@jziolkowski did not drop support of his program Tasmotizer. 
In everyones life are changes... Priorities have to or need to be changed.
Keep in mind it is done in free time and no one pays for.
Stay tuned. He is working on features requested here. It is done when it is done 😎

discip, 2020-09-01:
No no, please do not get me wrong. 😱
I did not say @jziolkowski dropped this project. But it rather seemed, that this was the fact.

I do appreciate all the people, who participate in opensource and do beautiful things like ***"Tasmotizer"*** .
And I, of course understand, that people have higher priorities than programming software.

So, I am willing to wait until it is finished.

jziolkowski, 2020-09-01:
Thanks for all the kind words ;) I completely agree that the project might have looked as abandoned, but it's really not. As I write this, there are new features being tested (available under 1.1d branch) to be released this week. And indeed @Jason2866 deserves the praise, as his stepping up to fix the OTA server issue gave me the needed kick to get back to Tasmotizer.

discip, 2020-09-01:
@jziolkowski
Oh no. I am the one, who should **thank you**, for developing such beautiful piece of software!
That's good news! But please do not push yourself and take your time.
I am used to go with these sayings: ***"A good thing takes time!"*** and ***"Good things come to those who wait!"***.