tasmota/tasmotizer #40: Config reliability

scargill, 2020-05-03:
No doubt about it, Tasmotizer is the best way to Tasmotise our ESPs, but on several occations I've found the CONFIG section less than reliable. If I use asmotizer then flick over to my phone and enter the SSID, pass and ID, the chip hops onto my network every time, but if I choose instead to use the config page on Tasmotizer, it doesn't always work and I sometimes end up having to re-flash the ESP (using Tasmotizer of course). Am I alone?

NicLogix, 2020-05-03:
Peter,

I found that setting too much info at once does not work with Tasmota.

I keep a couple of text lines, which I copy-paste in the console.

1.  Network settings ( ex. ssid1, password)
2.  Other network settings ( ex. Ntp, timezone )
3.  States and Mqtt settings (ex. poweronstate, ledstate, mqtthost,
mqttuser )
4.  Custom names ( ex. Friendlyname, grouptopic )

Tasmota wants to restart on some settings, configuring on baby steps seen
to work better for me.

Also I flash a blank, disconnect and leave off for an hour.  Then flash
(with erase) Tasmota.
Connect with terminal, do settings slowly.

Hope it helps.

Logix.

jziolkowski, 2020-09-21:
Please test the rewritten send dialog functions from branch develop-1.3