tasmota/tasmotizer #42: Bug: Tasmotizer stucks in progress after backup finished

golddragon007, 2020-05-20:
Sonoff products need a restart after the backup was made to be possible to erase and flash. However tasmotizer doesn't do it, and I need to do it, but at that moment and time the application crashes. And it's doesn't feel too well to randomly disconnect a device in the middle of theoretically flashing state.

ptz0n, 2020-07-12:
I was running into the same issue. First off, did you disconnect the IO100 before flashing? I tried first with a fixed jumper wire without success.

1. Solder a button or detachable wire to IO100.
2. Make sure the "Backup" checkbox is unchecked before flashing.

That worked perfectly for me on macOS.

Good luck.

Gosokan, 2020-07-14:
I was running into the same issue on Windows 10.  Sonoff POW R2 backup file size is only 1 MB instead 4 MB.

aserraric, 2020-07-18:
@ptz0n Well, if you don't make a backup, you are not going to run into the issue.

I had the same problem with two Sonoff S26 and an auvisio S06. I ended up making the backup separately and using Tasmotizer only for flashing and config. Windows 10

gadget78, 2020-08-16:
i also get this, does feel like its crashed even thou it does change from % done to... 
Writing Done
but the button doesn't change from "abort" and when you select that it gives warning "Flashing Aborted by user"
and then program closes ! 

it DOES give you a decent backup, of which it seems its about the only decent easy way todo ! 
but would be good to have it finish properly, give you an OK button, and say reboot device to continue flashing maybe etc etc

@aserraric what method/program did you make a backup seperately ?

aserraric, 2020-08-28:
> @aserraric what method/program did you make a backup seperately ?

I used the python script (esptool.py).

Jason2866, 2020-08-29:
It is a bug. Tasmotizer crashes because when the device is disconnected the COM port is not
available anymore and Tasmotizer does not respect this...
Until this is fixed. You can do a backup. Check `Backup`  The backup is fine (Tasmotizer crashes later). Connect device again, start Tasmotizer, for flashing uncheck Backup and hit Tasmotize

discip, 2020-08-29:
@Jason2866 
OK, sound fair enough.
For now.

I am reading out of :

> Until this is fixed.

that this is going to be fixed soon, right? 😁

thanks
regards

Jason2866, 2020-09-20:
Not completely last work done but ready for testing https://github.com/tasmota/tasmotizer/tree/1.1d

discip, 2020-09-20:
@Jason2866
That sounds promising!
Would you mind to provide *a compiled version for win10*?
thanks in advance

Jason2866, 2020-09-21:
v.1.2. is released. You find the win10 compiled version in releases

jziolkowski, 2020-09-21:
Closing b/c tasmotizer now supports devices without self reset, therefore this bug is fixed. I'll reopen if needed.

closed 2020-09-21 by jziolkowski