tasmota/tasmotizer #45: Can some one help the novice

hazelldrums, 2020-06-28:
I've installed python on my mac and run pip3 install tasmotizer on my Terminal.

But i get this screen when trying to open the program

tvfox:~ michaelhazell$ tasmotizer.py
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/tasmotizer.py", line 7, in <module>
    import tasmotizer_esptool as esptool
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/tasmotizer_esptool.py", line 36, in <module>
    from PyQt5.QtCore import pyqtSignal, QObject, pyqtSlot
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/PyQt5/QtCore.abi3.so, 2): Symbol not found: _LSCopyDefaultApplicationURLForURL
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore
  Expected in: /System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices
 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore

What am i missing here?

MPW1412, 2020-09-17:
Hi,

looks like you didn't install PyQt5. And you'll also need pyserial.

And in my case, I also had to install libxcb-xinerama0-dev

```
sudo pip3 install PyQt5 pyserial
sudo apt-get install libxcb-xinerama0-dev
```

feliperafael, 2020-11-19:
I had the same problem. I decided to install by "conda"

look at this issue: 
#https://github.com/tasmota/tasmotizer/issues/63

closed 2020-12-22 by Jason2866