tasmota/tasmotizer #46: OSX issue

podarok, 2020-07-16:
```
sudo tasmotizer.py
esptool.py v2.8
Serial port /dev/tty.usbserial-14610
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 86, in execute
    esptool.main(command)
  File "/usr/local/lib/python3.8/site-packages/tasmotizer_esptool.py", line 2930, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/usr/local/lib/python3.8/site-packages/tasmotizer_esptool.py", line 268, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/local/lib/python3.8/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 435, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
Abort trap: 6

```
When trying to Flash

podarok, 2020-07-16:
Tested latest master
```
python ./tasmotizer.py
esptool.py v2.8
Serial port /dev/tty.usbserial-14610
Traceback (most recent call last):
  File "./tasmotizer.py", line 86, in execute
    esptool.main(command)
  File "/private/var/www/tasmotizer/tasmotizer_esptool.py", line 2930, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "/private/var/www/tasmotizer/tasmotizer_esptool.py", line 268, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/local/Cellar/python@3.8/3.8.4/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/local/Cellar/python@3.8/3.8.4/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/usr/local/Cellar/python@3.8/3.8.4/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/serialposix.py", line 435, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
Abort trap: 6
```
same issue

[image]

podarok, 2020-07-16:
```
pip3 install PyQt5 pyserial
Requirement already satisfied: PyQt5 in /usr/local/lib/python3.8/site-packages (5.15.0)
Requirement already satisfied: pyserial in /usr/local/lib/python3.8/site-packages (3.4)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in /usr/local/lib/python3.8/site-packages (from PyQt5) (12.8.0)
```

raenrfm, 2020-08-20:
Not sure if it's related to your issue but seems like it, I get this with Linux Mint 20:

Traceback (most recent call last):
  File "tasmotizer.py", line 15, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'

I also have all the requirements satisfied.

thomo, 2020-08-29:
Same using High Sierra - when selecting checkbox "Release" or "Development" it crashes. I then check if I can reach "`thehackbox.org`" in the browser - and it can't be resolved. 

Whatever the reason, it should be so that if `http://thehackbox.org/tasmota/release/release.php` / `http://thehackbox.org/tasmota/development.php` are not available, appropriate error handling is done.

Tasmotizer 1.0
PyQt5 5.15.0
macOS 10.13.6

Jason2866, 2020-08-29:
Please use version 1.1c the OTA source has changed

thomo, 2020-08-29:
I rechecked it (with 1.0) after thehackbox.org is reachable again - and it work now as expected.

closed 2020-08-29 by Jason2866