tasmota/tasmotizer #47: Flashed but wifi does not stay on long enough to connect

Dutchy-79, 2020-08-09:
Hi,

I just flashed a TYWE3S and it said it was done successful.
I disconnected the wires and put the batteries back in, My phone finds the wifi network, It connects without internet and then when I enter the IP 192.168.4.1 in my chrome browser the wifi disconnects and can't be found up-until I take the batteries out and insert them again. Then same thing happens.

I flashed Tasmota.bin to a tuya wifi door sensor using tasmotizer.  OTA did not work.

Please give me some advice

Dutchy-79, 2020-08-09:
Add, I did try 2 new pairs of AAA's to see it wasn't low power.

Dutchy-79, 2020-08-09:
After several attemps and being ready with everything filled out in the browser I did manage to setup wifi.
Now the problem is still there, It keeps losing connection every few seconds.

> 

Program Version | 8.4.0(tasmota)
-- | --
2020-07-29T12:05:28
2_7_2_1/2.2.2-dev(38a443e)
0T00:00:11
43 at 0xF9000
38
Power On
Tasmota
 
wifi_smart (46%, -77 dBm)
tasmota_A4B114-4732
84:0D:8E:A4:B1:14
192.168.1.156
 
192.168.1.1
255.255.255.0
192.168.1.11
 
 
1883
DVES_USER
DVES_A4B114
tasmota_%06X
cmnd/tasmotas/
cmnd/tasmota_A4B114/
cmnd/DVES_A4B114_fb/
 
None
Disabled
 
10793236
0x1540C8
2048kB
1024kB
592kB
408kB
27kB

jziolkowski, 2020-08-09:
Tasmotizer is only for flashing. For the actual Tasmota support post an issue on tasmota github or (better) ask on the official Discord server https://discord.gg/Ks2Kzd4

closed 2020-08-09 by jziolkowski