tasmota/tasmotizer #52: tasmotizer_x86-1.1c.exe gives "Failed to execute script" on startup

Sum-Guy, 2020-09-12:
I' trying to run tasmotizer_x86-1.1c.exe (as admin) on Win-7 32 bit and all I get is a message window saying "Failed to execute script Tasmotizer". 

Any ideas what's wrong?

Sum-Guy, 2020-09-12:
Do I have to install any sort of python thingy for this so-called "stand-alone" tasmotizer to work?

The source-code zip file contains a requirements.txt that contains only this one line:

pyserialPyQt5>=5.10

what the hell is that?

Sum-Guy, 2020-09-12:
On a different win-7 32-bit system I have, tasmotizer_x86-1.1c.exe DOES run (and does not appear to need run-as-admin to run).    I've looked at the list of running services (command shell - net start) and there were a few services I had turned off that I enabled (and started) and still I can't run it on the first system (failed to execute script).

On the second system there is some visual studio stuff installed, and for all I know maybe some different dot net stuff that might not be the same?

Sum-Guy, 2020-09-13:
On the system where I can't run the stand-alone tasmotizer_x86 exe, turns out I do have python installed, and I just did install this:  PyQt5-5.15.1-5.15.1-cp35.cp36.cp37.cp38.cp39-none-win32.whl.  That still doesn't help me run the stand-alone exe (same error).   When trying to run the tasmotizer.py from a command line, I get "module not found error.  No module named 'serial'.  There's also a setup.py but I get "unhashable type: 'dict'".

Jason2866, 2020-09-20:
Sorry cant help. In the PC which it doesnt run "something" is screwed

closed 2020-09-20 by Jason2866