tasmota/tasmotizer #53: Tasmotizer 1.1c for esp32

geyseran, 2020-09-16:
I am trying to flash tasmota32-webcam.bin to esp32 device, but the Tasmotizer does not see the tasmota32 firmware files. I downloaded the tasmota32-webcam.bin file and tried to use Tasmotizer 1.1c to flash the device, but I get an error "Invalid head of packet(0x46)". Is there a Tasmotizer version that would flash the tasmota32 files?

Jason2866, 2020-09-20:
No. There is no plan for a ESP32 version

jziolkowski, 2020-09-20:
... for the time being. After ESP32 is stable in Tasmota, it will be made possible.

closed 2020-09-21 by jziolkowski

sanchosk, 2021-09-01:
I think we are there already - ESP32 has daily builds and works OK. Any plan to include it?

Jason2866, 2021-09-01:
Do not expect anything. This is a major rewritte. Jacek (the author) does not have (free) time to do.
Flashing the ESP32 is easy possible (without the features Tasmotizer provides!) with a Chrome or Edge browser via Tasmota [WebInstaller https://arendst.github.io/Tasmota-firmware/](https://arendst.github.io/Tasmota-firmware/)

sanchosk, 2021-09-01:
Got the message. I've looked into the tasmotizer_esptool and it seems to already support all the calls for ESP32, including the binaries necessary to load the firmware.
Not a big python pro, but I'll try to check if I can mod it myself, I can then propose the pull request.

Jason2866, 2021-09-01:
A PR is very welcome! And yes the included esptool.py is just a bit enhanced version of the official esptool.py. If you are "on it" please update the esptool.py version to 3.1.
AND you have opened a can of worms 😉 Flashing the ESP32-C3 is the next wanted feature.
Tasmota runs already very stable with