tasmota/tasmotizer #55: Tasmotizer not starting

NicBout, 2020-09-21:
Hi,

Tasmotizer stopped working for me today:

I tried upgrading to version 1.2 and I get this error in Terminal:
tasmotizer.py
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 23, in <module>
    from utils import MODULES, NoBinFile, NetworkError
ModuleNotFoundError: No module named 'utils'

I use a Mac running the latest version of Catalina.

Any idea?

Many thanks!

N

jziolkowski, 2020-09-21:
using the pip build?

NicBout, 2020-09-21:
Yes:
I used "pip3 install tasmotizer" to run the installation.
Tried installing after a restart too.

jziolkowski, 2020-09-21:
Yes, I see some issues as well
damnit. on it.

NicBout, 2020-09-21:
I get this:

Installation:
pip3 install tasmotizer
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: tasmotizer in ./Library/Python/3.8/lib/python/site-packages (1.1.1)
Requirement already satisfied: PyQt5>=5.10 in /Library/Python/3.8/site-packages (from tasmotizer) 
(5.15.1)
Requirement already satisfied: pyserial>=3.0 in /Library/Python/3.8/site-packages (from tasmotizer) (3.4)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in /Library/Python/3.8/site-packages (from PyQt5>=5.10->tasmotizer) (12.8.1)

Runtime:
tasmotizer.py
Traceback (most recent call last):
  File "/usr/local/bin/tasmotizer.py", line 23, in <module>
    from utils import MODULES, NoBinFile, NetworkError
ModuleNotFoundError: No module named 'utils'

Am I doing something wrong?

jziolkowski, 2020-09-21:
try these:
`pip3 install --upgrade pip wheel tasmotizer`

jziolkowski, 2020-09-21:
I've created a new empty venv, upgraded pip and wheel, installed tasmotizer
installed ver. 1.2.1, works as expected

NicBout, 2020-09-21:
Installed successfully now,
also "tasmotized" a test device and tried a simple send config. 
All ok...

Thanks a lot.

jziolkowski, 2020-09-21:
Glad to hear :)

Based on your experience I've updated the installation steps in the readme.

closed 2020-09-21 by jziolkowski