tasmota/tasmotizer #59: Policy Kit Authentication

ShayBox, 2020-10-06:
Instead of having to run tasmotizer in a terminal with sudo, built-in polkit auth would allow any user with sudo access to enter their password in a prompt when the program is run, this allows for use of a .desktop entry.

This can be done many ways internally, the easiest would be something like wraping it in [pkexec](https://www.freedesktop.org/software/polkit/docs/0.108/pkexec.1.html), but that would be the bare minimum, the more common way would be to split action(s) that requires root access into a separate runnable file, then run that action with pkexec, this allows the password prompt to show when you go to do the action instead of every time the program opens, this does require the machine have a polkit authentication agent installed ([most desktop environments do](https://wiki.archlinux.org/index.php/Polkit#Authentication_agents))

jziolkowski, 2020-10-06:
why do you need sudo to run tasmotizer? simply add your user to `serial` group and be done with it

ShayBox, 2020-10-06:
I prefer not to add every group to my user to bypass authentication prompts, in this case serial is harmless, I don't use my serial ports, but I don't like every program having access to my hardware.

The bad case is if you give your user the `disk` group, every program is able to do anything to every disk without any notification or authentication, format, write data, erase data, anything.

jziolkowski, 2020-10-06:
Ah so it boils down to how stupid Arch is.

I'm open to PR as I won't be even able to test it. If you can't provide one then sorry.

ShayBox, 2020-10-07:
It's not just an arch thing, many distros have been removing groups that give users access to hardware because its a huge security flaw.

Assuming you're using any desktop environment, you would have a polkit agent, many, many programs use it, you would likely run into problems using programs if you didn't.

I'm down to implement it in a pr, but I wanted to get maintainers to talk and see how they'd prefer it to be implemented.

jziolkowski, 2020-11-11:
Closing due to inactivity

closed 2020-11-11 by jziolkowski