tasmota/tasmotizer #62: ESP Wroom 32 error "Invalid head of packet"

bertiebeez, 2020-10-11:
I am trying the tasmotizer for the first time on a espwroom 32 board. I selected bin file tasmota32.bin and com 4. I get a quick popup that say's it is "connected to esp8266ex", then another popup that fails due to "invalid head of packet".
I can flash this with Arduino IDE by holding down the boot button and waiting until I see "Connecting..." then I release the button and it will flash ok. So I know the com port and usb cable are good. Baud rate could be a problem??
Thanks for any help.

jziolkowski, 2020-10-11:
Tasmotizer doesn't support ESP32 yet.

bertiebeez, 2020-10-12:
Thanks for the reply. What is the difference in the tasmota.bin versus the tasmota32.bin? Sent from my Samsung Galaxy Tab®|PRO

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/tasmota/tasmotizer/issues/62#issuecomment-706733002",
"url": "https://github.com/tasmota/tasmotizer/issues/62#issuecomment-706733002",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

jziolkowski, 2020-10-12:
It's not the .bin difference, but it's a completely different device. The good thing is that esptool (which tasmotizer uses) supports ESP32, so I need to add ESP32 support to tasmotizer as well. But that will happen not earlier than next month.

bertiebeez, 2020-10-12:
I tried to load the source file from github into arduino idea and if failed for numerous missing files during the compile. I then tried the tasmatizer and it gave the error that I mentioned before. I did get the nodemcu py programmer to work just fine and have the tasmota32.bin flashed on the esp32 and it seems to work with a couple of test sketches.Sent from my Samsung Galaxy Tab®|PRO

closed 2020-11-11 by jziolkowski

ayavilevich, 2020-11-11:
Not going to happen?

jziolkowski, 2020-11-11:
It will happen eventually when ESP32 in Tasmota is not in beta anymore. Until then use bare esptool

scargill, 2021-08-17:
August 2021 I'm getting that "invaalid head of packet (0x46)" -any ideas?? I'm using tasmota32.bin

kirax999, 2021-09-05:
jai le meme souci

Darthmonkey, 2023-06-12:
I assume that as I'm getting this issue while in June 2023, that this isn't going to happen now?

ayavilevich, 2023-06-12:
I believe the preferred tool now is the Tasmota Web Installer https://tasmota.github.io/install/

Cysign, 2024-11-21:
Still no news oin Tasmotizer? I really liked this tool.
The web-tool doesn't work properly anytime... sometimes it works, 20 other times it doesn't work. Have been trying different cables...but the behavior always is the same...