tasmota/tasmotizer #64: Flash only works if download firmware option is enabled

bakoontz, 2020-11-10:
I am flashing a couple of Sonoff basic switches, RF R2 v1.0 per the instructions on the tasmota.github.io site.  I am using an Arduino UNO for the USB to TTY conversion per these instructions:

https://siytek.com/how-to-flash-sonoff-tasmota-with-an-arduino-uno-and-esptool/

The update consistently failed when Backup: Save original firmware was selected (along with Release and Erase), usually with byte errors.  Disabling Backup allows the flash update to progress normally.

Downloading the firmware isn't really that important to me, I'm very pleased with the new firmware.  I just thought you all might want to know.

jziolkowski, 2020-11-11:
Which version has this issue?
Also, you don't have to download the firmware each time as you can select a BIN file from your PC

I don't really follow what's the issue there

bakoontz, 2020-11-11:
See attached image.  This option, when enabled, causes the session to fail.  When it's disabled, all works well.

[image]

jziolkowski, 2020-11-11:
Do you power cycle the device after backup?

bakoontz, 2020-11-11:
Never gets that far, the backup fails almost immediately, usually with byte mismatch errors.  Disabling the backup option does allow the flash to continue to completion.

jziolkowski, 2020-11-11:
Interesting. Can you make a video of the process? Or at least post a screenshot of the error? Preferably the console output as well (run it inside cmd.exe)

Jason2866, 2020-11-11:
Ouuch. Very bad idea to use a device with 5V logic levels.
It is just **WRONG** that the ESP8266 has GPIOs which are 5V tolerant.
Weird things can happen when the ESP8266 is tortured with 5V. You encountered one of.
Using the recommended USB serial adapter for flashing a ESpxxxx will result in a successfull flash

closed 2020-11-11 by Jason2866

jziolkowski, 2020-11-11:
Dang indeed. When reading the original issue somehow I omitted it's an UNO and assumed you are using the 3v3 version.
Then Jason is correct, this issue is not related to Tasmotizer.

bakoontz, 2020-11-11:
That's why I mentioned it in my initial writeup.  There is ample evidence that ESP GPIOs are 5V-tolerant.  But it's whatever, still happy with the Tasmotizer.

jziolkowski, 2020-11-11:
If it were, then your little setup would work. ;)

bakoontz, 2020-11-11:
It works great actually.  I don't need Sonos' crappy firmware.  But yes, a proper USB-to-TTL adapter is a good suggestion.

bakoontz, 2020-11-13:
@Jason2866 @jziolkowski Using a proper USB-to-TTL converter did the trick.  Thanks!

L4RGER, 2020-12-04:
> Ouuch. Very bad idea to use a device with 5V logic levels.
> It is just **WRONG** that the ESP8266 has GPIOs which are 5V tolerant.
> Weird things can happen when the ESP8266 is tortured with 5V. You encountered one of.
> Using the recommended USB serial adapter for flashing a ESpxxxx will result in a successfull flash.
> It is just that easy. Case closed.

I’m having the same issue using 3.3v;
I’ve flashed ~12 or so devices and not been able to back up original firmware on; I’m using a PL2302 USB to TTL.

bakoontz, 2020-12-04:
There’s a timing issue involved: Try starting your download IMMEDIATELY
after applying power. It took a couple tries for me to get it to work.

L4RGER, 2020-12-04:
Thanks mate, will do

Jason2866, 2020-12-05:
We do recommend for a worry free flashing "the golden CH340" https://tasmota.github.io/docs/Getting-Started/#needed-hardware

jziolkowski, 2020-12-05:
as Jason said, ESP chips from one batch can be more/less susceptible to quality of USB adapters. CH340G is the recommended variant in regards to power supply, operational stability and stuff like that. Myself I have a PL2302 as well and it gives mixed results, whereas CH340G has a 100% success rate.

L4RGER, 2020-12-06:
You’ll both be please to know that I have got one on the way. I couldn’t source one locally so it’s on the slow SS AliExpress
I’m just not sure what the deal with the jumper between 3.3v and tx in the pictures is for but I will sort it out when (if) it arrives 
[image]

Jason2866, 2020-12-06:
The jumper placing is just for parking the jumper. Please remove and place it on the 5V pin.
Doing this it prevents you to accidently connect the 5v rail to the esp