tasmota/tasmotizer #65: Tasmotizer do not start on macOS Big Sur

typepub, 2020-11-21:
Hello,

Facing a problem on a fresh install of Tasmotizer:
python3 -m venv venv2
. venv2/bin/activate 
pip install tasmotizer
Successfully installed PyQt5-5.15.1 PyQt5-sip-12.8.1 pyserial-3.4 tasmotizer-1.2.1

When starting Tasmotizer, no window appear and Python is stuck 100% of CPU. Have to kill it. Tried to wait for 5 minutes but no window appear and still stuck.

Any idea of what I can try please?

Thanks.

jimmyhall82, 2020-11-23:
I am having the same issue, did you manage to solve this yet?

typepub, 2020-11-23:
No solution found. I used a workaround: https://tasmota.github.io/docs/Getting-Started/#esptoolpy

grobian19, 2020-11-24:
Same problem here

Jason2866, 2020-12-07:
Use latest pyserial for Big Sur see https://github.com/pyserial/pyserial/issues/509#issuecomment-726966076

closed 2020-12-07 by Jason2866

apastuszak, 2021-02-18:
The latest pyserial does not fix the issue.  Can we reopen this?