tasmota/tasmotizer #66: Would some one  please tell us non-Programmers were the backup .bins are kept can Details be that be put in plain sight

mutantxxxx, 2020-11-21:
Would some tell us non-Programmers were the backup .bins are kept can that be put in plain sight 
You mention backs made before flashing which it did
But were??? the are they were???  is the default place what are the default names ???
I could see that it was backing up but spent hour and gave up trying to find the backups when I wanted flash back original image .
So easy to do the authors put WERE THESE backup images wopuls saved ????
Yes I know we all have lots of time at present but doing full scans of c: drive 
to locate backup files  oohh dear come on gentalmen please a bit more info please . for non-programmers

ascillato, 2020-11-21:
Please, calm down!

It is stored as _backup_date_time.bin_ (like _backup_20201121_203138.bin_) in **the same folder** where you have put **_tasmotizer.exe_**

closed 2020-11-21 by ascillato

dartrax, 2021-01-07:
In my case, the files are not saved where the tasmotizer.exe is.
It get's saved in the folder of that exe that starts the tasmotizer.exe-process. That's a little bit annoying, but I admit that is a special case not common to everyone.

jziolkowski, 2021-01-07:
Sorry, but I don't get the difference between where tasmotizer is and the other location you mentioned

dartrax, 2021-01-07:
Thanks for the replay!
Tasmotizer is here:
D:\Projects\TASMOTA\Tasmotizer\tasmotizer-1.2.exe
The backups are saved here:
D:\Drive\OrderManager\bin\Debug\

jziolkowski, 2021-01-07:
ahhh ok now I get it.
Then you are correct. The bins are saved in the _working_ directory, not exactly the same where tasmotizer.exe is

mac-city, 2021-04-02:
what about us mac users?  can't find my backup anywhere :-(  plus it is crashing the gui after backup but I am hoping the backup completed somewhere so I can try to restore my sonoff to original firmware.