tasmota/tasmotizer #67: Access to USB port

silentnelis, 2020-12-18:
I’m a beginner and need some help, I try to flash Sonoff S20 device through Tasmotizer 1.2.1 on a mac (Big Sur) but one way ore the other the Tasmotizer doesn't see the usb port after refreshing. Driver installed and hardware USB device Tree even shows FT232 USB UART. 
A strange thing is that if i'm looking at Hardware at the USB Device tree I see that FT232 USB UART showing in the USB 3.0 Bus section while my Wacom intuos pro connection is in the USB2.0 Hub section.
Any ideas?

silentnelis, 2020-12-18:
Additional pictures
[image]

Jason2866, 2020-12-22:
It is a issue from Big Sur. Update to latest development version of pyserial.
https://github.com/tasmota/tasmotizer/issues/65

jziolkowski, 2020-12-22:
Post a reply if what Jason2866 suggested worked, then the issue will be closed as fixed.

closed 2020-12-22 by Jason2866