tasmota/tasmotizer #70: Friendly name

scargill, 2021-01-02:
I just realised an issue with the config section of the otherwise excellent Tasmotize (1.2 - I'm using the 64 bit Windows binary).  In "send config" when settingthe "friendly name" - this is NOT setting the hostname... 

So for example - I set up a new nodeMCU board - called it "nodemcu-4" in the config. When I tried to test the board in the Tasmota webUI, nothing - so I did a status 5 request in MQTT - sure enough that worked - the name works for the MQTT topic... but in the webUI - the board is called nodemcu-1-xxxx where xxxx is a 4-digit number.  That can be manually fixed but I'll bet that's caused loads of confusion. 

[image]

Any change of an update?

jziolkowski, 2021-01-02:
Hi Pete,
well this is works as expected. It sets the FriendlyName, not hostname. But since I should update the config to set DeviceName anyway, I can add the hostname field as well.

scargill, 2021-01-03:
That would be marvellous, thanks.

scargill, 2021-01-28:
In configuration -  if I make for example a setup including friendlyname display1 (I just did that to test) I end up with wifiname friendlyname1-1030 (or similar) and in Tasmota config - other - I get friendlyname1=display1  but devicename is tasmota.

It seems to me that in the configuration even if you feel these should be separate, we should be able to set all three. The WiFi in particular is confusing - you have to go find the device by IP (there's a get IP but that never works for me) instead of just finding the device by the name just given. Any chance? Really would be appreciated.

Pete

scargill, 2021-02-14:
Any chance of that friendly name fix? Every time I flash tasmota I have to go back in and fix the name?