tasmota/tasmotizer #71: PyQt 5.15 issue with Debian Buster 10.7

Mrnt, 2021-01-06:
### Issue

Installed Tasmotizer in a virtual environment on Debian Buster 10.7 using the "Option 2" instructions:

> Option 2: Install from PyPI using pip:
> - Upgrade pip and wheel packages: pip3 install --upgrade pip wheel
> - Install tasmotizer: pip3 install tasmotizer
> - Then simply run tasmotizer.py from the shell/command line
 
When running Tasmotizer on Debian Buster the app fails to start - the output includes the following error:

```
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

Setting `export QT_DEBUG_PLUGINS=1` then running Tasmotizer again gives some additional information:

```
Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/-path_to_venv-/bin/platforms" ...
Cannot load library /-path_to_venv-/lib/python3.7/site-packages/PySide2/Qt/plugins/platforms/libqxcb.so: (libxcb-util.so.1: cannot open shared object file: No such file or directory)
QLibraryPrivate::loadPlugin failed on "/-path_to_venv-/lib/python3.7/site-packages/PySide2/Qt/plugins/platforms/libqxcb.so" : "Cannot load library /-path_to_venv-/lib/python3.7/site-packages/PySide2/Qt/plugins/platforms/libqxcb.so: (libxcb-util.so.1: cannot open shared object file: No such file or directory)"
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

```

### Cause

It appears that as of PyQt5 5.15, the Qt plugin `libqxcb.so` is expecting to find the module `libxcb-util.so.1` eg:
`/lib/x86_64-linux-gnu/libxcb-util.so.1`
However, Debian Buster (version 10.7) only has `libxcb-util.so.0`
This change was apparently made in PyQt 5.15. 

### Possible Solution

One possible solution might be to pin the max requirement for PyQt5 to 5.14.2 though I have not tested this (I opted to make a symbolic link called `libxcb-util.so.1` in the same folder as `libxcb-util.so.0` that pointed to `libxcb-util.so.0`, which seems to work).

jziolkowski, 2021-01-06:
Oh that's interesting, as it actually might finally be a solution to many XCB related issues posted! Thanks! I'll forward other users who posted this issue to this one so we can verify!

Jason2866, 2021-03-04:
Requirements.txt changed to 5.14

jvw1954, 2021-03-10:
`sudo ln -s /usr/lib/x86_64-linux-gnu/libxcb-util.so.0 /usr/lib/x86_64-linux-gnu/libxcb-util.so.1`
solved my problem

houldsg, 2022-01-14:
I got this same error on Ubuntu 20.04 and resolved it by running  ```sudo apt install libxcb-xinerama0```

algspd, 2022-02-05:
> `sudo ln -s /usr/lib/x86_64-linux-gnu/libxcb-util.so.0 /usr/lib/x86_64-linux-gnu/libxcb-util.so.1` solved my problem

This is the solution in debian 10.11

Vlad0zz, 2022-09-29:
sudo ln -s /usr/lib/x86_64-linux-gnu/libxcb-util.so.0 /usr/lib/x86_64-linux-gnu/libxcb-util.so.1 solved my problem

ASTRA LINUX