tasmota/tasmotizer #73: "send Config" won't work (Ubuntu 20)

adizag, 2021-01-09:
Unable to send config after flashing. nor get an IP address (shows xx.xx.xx.xx).
getting a message from Tasmotizer but nothing happens.

[image]

jziolkowski, 2021-01-10:
The problem with send config and IP read is that there is not 100% reliable way of checking device state after flashing.
Which device are you using? What settings?

OfferLa, 2021-01-10:
For me, Sonoff Basic, Windows 10, Tasmotizer is flashing ok, can send SSID by 'send config', but never succeeded to get ip by tasmotizer query button.

jziolkowski, 2021-01-10:
Do you restart and/or power cycle the device after flashing?

adizag, 2021-01-11:
@jziolkowski 
I'm on Ubuntu 20, installed both with method 2 and 3 in the readme.
flashing sonoff T1.
tried power cycle several times.

I have some problems with pre-required python packages at the beginning (showed as an error in terminal) but I can't see any similar message now.

pafribouf, 2021-06-14:
Workaround: use esptool

sausin, 2021-09-02:
Having the same issue as @adizag. 

Send config was done after power cycling and removing connection to GND. However, in all trials so far it didn't work either with ESP8266-01 or ESP8266 12-E NodeMCU