tasmota/tasmotizer #75: tasmotizer - Multi language support

bovirus, 2021-01-16:
@Jason2866 

Using PO language file and xgettet it's possible to localize tasmotizer in several languages.

I already tested that it works.

If you are interested I can put all chanegs about it (also ina development branch)

It's the same system that we used to localize "DDT4ALL" github project

Use a "tools" folder where you can make the batch to create the template language file and align language files to template

Use a "local\xx" (where xx can be  DE or IT or FR) subfolder where you can have the language files.

In .py Pyhton files replace es. 

print ("Test") 

with 

print (_("test") 

and "Test" string will be added to template file and can be localized.

jziolkowski, 2021-01-16:
make a PR and I'll have a look

jziolkowski, 2021-01-16:
please base it on develop-1.3 branch

bovirus, 2021-01-16:
@jziolkowski 

OK: I will make a PR with general tool and some changes on basic string for GUI (Ex Serial port / etc).

Then you can check if it' works fine and we can go away.

Is it Ok for you?

jziolkowski, 2021-01-16:
Sure :)

bovirus, 2021-01-16:
@jziolkowski 

The you confirm that I will make a PR n dev 1.3 also is less updated then release on github repository?

jziolkowski, 2021-01-16:
if you compare changes these are some very minor things. Make the PR anyway, I will take care of these differences when rebasing prior to new release.

bovirus, 2021-01-16:
Ok.

I will do as soon as.

bovirus, 2021-01-17:
@Jason2866 

Done

https://github.com/tasmota/tasmotizer/pull/77

bovirus, 2021-02-14:
Any option to continue with the localization job?

bovirus, 2021-04-02:
Are there any plan about multilanguage GUI?

Jason2866, 2021-04-02:
Next planned feature is adding ESP32 support. Not saying there is a time schedule for.
Time is very rare for spending time on. If this is done multilanguage support is considered