tasmota/tasmotizer #76: PyQt5 serial port support isn't default on Ubuntu

freedomlives, 2021-01-17:
First I installed on my Gentoo desktop, and got a similar error, but using Gentoo is my own choice to be masochistic and see what all needs to be compiled a given way. However, I now installed on my Ubuntu laptop and got the same error.
Given the ubiquity of Ubuntu users, might be good to add to the install instructions that they need to: sudo apt install python3-pyqt5.qtserialport

$ tasmotizer.py 
Traceback (most recent call last):
  File "/home/andrew/.local/bin/tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'

jnth, 2021-10-30:
You cant use `pipx` to install `tasmotizer`. It works fine (tested with the last version of Ubuntu).
But, uninstall all previous installation of `tasmotizer` before (to avoid confilct).

```
pipx install tasmotizer
```

Richileaks, 2023-04-03:
Hi,
i use ubuntu 22.04.2 LTS

~/delme (venv) 
$ pip3 install tasmotizer
Collecting tasmotizer
  Using cached tasmotizer-1.2.1-py3-none-any.whl (196 kB)
Collecting PyQt5>=5.10
  Downloading PyQt5-5.15.9-cp37-abi3-manylinux_2_17_x86_64.whl (8.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.4/8.4 MB 12.4 MB/s eta 0:00:00
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting PyQt5-Qt5>=5.15.2
  Using cached PyQt5_Qt5-5.15.2-py3-none-manylinux2014_x86_64.whl (59.9 MB)
Collecting PyQt5-sip<13,>=12.11
  Downloading PyQt5_sip-12.11.1-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.whl (360 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 360.1/360.1 KB 10.8 MB/s eta 0:00:00
Installing collected packages: pyserial, PyQt5-Qt5, PyQt5-sip, PyQt5, tasmotizer
Successfully installed PyQt5-5.15.9 PyQt5-Qt5-5.15.2 PyQt5-sip-12.11.1 pyserial-3.5 tasmotizer-1.2.1

~/delme (venv) 
$ tasmotizer.py 
Traceback (most recent call last):
  File "/home/richi/.local/bin/tasmotizer.py", line 16, in <module>
    from PyQt5.QtSerialPort import QSerialPortInfo, QSerialPort
ModuleNotFoundError: No module named 'PyQt5.QtSerialPort'

if i run
$ python3 /home/richi/.local/bin/tasmotizer.py
does it work