tasmota/tasmotizer #82: Can't start Tasmotizer (freshly installed Debian Buster)

Pascal1Schneider, 2021-01-27:
[image]
As you can (hopefully) see in the Picture above, i installed Tasmotizer and tried to execute it, but there is always this error message and it's not only me some friends have the exact same problem, but we couldn't find a solution

Pascal1Schneider, 2021-01-27:
I found a solution to that problem, debian buster 10.5 has renamed some files/directories, so i created a symlink to libxcb-util.so.0

here is the command for it: ln -s libxcb-util.so.0 libxcb-util.so.1

JeremyDWilliams, 2021-03-01:
Same issue. Using  ln -s libxcb-util.so.0 libxcb-util.so.1 as shown above did not work for me.

Running 10.13.6 and based on error log, maybe needs 10.15 or higher?

log:
Traceback (most recent call last):
  File "PyInstaller/hooks/rthooks/pyi_rth_multiprocessing.py", line 17, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 531, in exec_module
  File "multiprocessing/__init__.py", line 16, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 531, in exec_module
  File "multiprocessing/context.py", line 6, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 531, in exec_module
  File "multiprocessing/reduction.py", line 16, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 531, in exec_module
  File "socket.py", line 54, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 531, in exec_module
  File "selectors.py", line 12, in <module>
ImportError: dlopen(/var/folders/cb/c2b5jrk50xnfm6xy3z0dl2kr0000gn/T/_MEIazdmo6/select.cpython-39-darwin.so, 2): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/cb/c2b5jrk50xnfm6xy3z0dl2kr0000gn/T/_MEIazdmo6/select.cpython-39-darwin.so (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/cb/c2b5jrk50xnfm6xy3z0dl2kr0000gn/T/_MEIazdmo6/select.cpython-39-darwin.so
[49364] Failed to execute script pyi_rth_multiprocessing

jziolkowski, 2021-03-02:
did you try #71 ?

jvw1954, 2021-03-10:
> Same issue. Using ln -s libxcb-util.so.0 libxcb-util.so.1 as shown above did not work for me.
Try:
`sudo ln -s /usr/lib/x86_64-linux-gnu/libxcb-util.so.0 /usr/lib/x86_64-linux-gnu/libxcb-util.so.1`