tasmota/tasmotizer #83: No ports are showing.

kaman1990, 2021-01-27:
No COM ports are showing when trying to use Tasmotizer, 

- Windows 10 - tried rebooting, plugging in different devices etc
- tried 32bit & 64bit
- tried run as administrator

[https://i.imgur.com/e791OYY.gif](url)

Any help appreciated.

jziolkowski, 2021-01-27:
Are the ports visible in the device manager?

jziolkowski, 2021-03-02:
Closing due to inactivity

closed 2021-03-02 by jziolkowski

chip-edw, 2021-05-17:
I know this is closed but I hope this helps someone.  Seems ports Com and lpt are legacy in windows 10 and not installed by default. I found the following Microsoft solution when I was searching and it helped me.
https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/how-to-identify-com-ports-in-windows10/2591ed8b-805e-4e66-9513-836cdd49ed80

aris-1, 2024-09-01:
I had to install these drivers
https://www.pololu.com/docs/0J7/all#2