tasmota/tasmotizer #84: Mac version hangs and have to  force quit

rlust, 2021-02-04:
The app loads but hangs on Big Sur.

antiguasky, 2021-02-07:
I'm experiencing the same behavior on Big Sur.    It loads but is not responding and I have to force quit.

moonberry-pl, 2021-02-15:
me to on Mojave 10.14.6 :(

andreasloe, 2021-02-20:
The same on 11.2.

finch6, 2021-02-27:
I'm having the same issue: Big Sur 11.3 Beta

cyclobob, 2021-02-28:
Same here, hangs on Big Sur 11.01 and need to force quit

fabiosoft, 2021-03-01:
Same for me

Jason2866, 2021-03-11:
Tested with VmWare on Windows 10 running MacOS Catalina.
Tasmotizer is working 
[image]

Jason2866, 2021-03-11:
Confirmed. Compile App is not working. The Python version is ok.
I have removed the compiled App.

finch6, 2021-03-11:
> Confirmed. Compile App is not working. The Python version is ok with MacOS Catalina. Can not test with BigSur
> I have removed the compiled App.
> Can you try if [ESP-Flasher App](https://github.com/Jason2866/ESP_Flasher/releases) is working with BigSur?
> It is tested with MacOS Catalina

The app runs but I don't have anything to flash at the moment to test if the flash actually works - but the app does run without error.

I'm still running MacOS BigSur 11.3 Beta

DCJones11, 2021-04-17:
I'm running Big Sur 11.2.3. Terminal just hangs after using the command tasmotizer.py. Need to quit Terminal to get out of it hanging.

GeorgeWithers, 2021-04-18:
Opening tasmotizer.py in the python IDLE and run it from there worked for me on 11.2!

andreasloe, 2021-08-01:
I think this can be closed due to the webpage that is available for flashing.