tasmota/tasmotizer #87: Apple Silicon / ARM

haakonstorm, 2021-02-17:
[image]

Tasmotizer has issues on the new ARM architecture Apple has switched to. This is on a MBA 13" M1. I first tried installing via `pipx` but that failed, so downloaded the release instead and that resulted in the above. Switched back to my macOS Catalina MBP (Intel) and Tasmotizer worked instantly, flashing a couple Sonoff Basic R2's. I just wanted to start this thread to follow up on status on (eventual?) ARM64 ("Apple Silicon") support for the 'tizer. Thank you all for your awesome work.

jziolkowski, 2021-03-02:
Yikes. Thanks for the heads up.

haakonstorm, 2021-03-03:
Stay awesome!

cwilmo, 2021-03-08:
I had the same problem on M1 Mac Mini but found a solution and wanted to share.  It works with Windows 10 ARM64 (Build 21327) under Parallels 16.3.2 Technical Preview using FTDI FT232R USB UART using the beta ARM64 drivers from FTDI.  The beta drivers (CDM v2.12.32.B3 for ARM64.zip) can be downloaded from ftp://ftp.ftdichip.com (user: u45902898-ide, pass: Ftd1$erv). I used the following steps:

There are 2 items of the drivers that needs to be installed, first the bridge, and after the serial to usb driver, both updated with inf from package from FTDI.
1. in an elevated command prompt (run as admin), disable the driver signature verification (test mode): 
bcdedit /set testsigning on
2. restart computer, you will notice Test mode in lower right corner
3. insert the adapter
4. after detecting it with exclamation sign: update driver, have disc -> point to the downloaded extracted driver in ARM64\Release folder; repeat for the serial to com driver

Thanks to ZoltanMarosan on Microsoft Community for the instruction.

I used Tasmotizer 1.2.exe and worked great.

haakonstorm, 2021-03-11:
Thanks for sharing!
I'll be watching this space for a native, non-beta solution meanwhile.

(It's really incredible how complex we humankind have made the process of flashing a few hundred kbs to a chip. Its in the same category as sending a large file over the internet without any hurdles.)

mbspears, 2021-07-08:
Any update on plans for a native version of Tasmotizer for M1 Macs?

jziolkowski, 2021-07-08:
@mbspears I have zero experience with native Mac apps, so most likely no.

mbspears, 2021-07-08:
Thanks for the update.  I completely understand.  Would have been nice if Apple made a backward compatible mode or something.

jziolkowski, 2021-07-08:
The additional issue is that I have no one nearby who has the M1 version, so I can't even properly debug where the issue is.

M1LLZR, 2021-07-08:
Has anyone got a good link to the (CDM v2.12.32.B3 for ARM64.zip) file? The server seems to have changed its password

Alexandtr1, 2021-07-16:
Hello  .I need driver ftdi .please send me to mail alexandtr1@mail.ru.

Alexandtr1, 2021-07-16:
Hello.
I need driver ftdi for programmer windows 10 on arm.
Send me files to alexandtr1@mail.ru

Noenemy88, 2021-08-13:
I lost driver for arm. please send the driver. Thank you.
dhwi12344.kim@gmail.com

ViroAR, 2021-08-17:
Hi, i need the driver please. santir.1098@gmail.com

smokescreen42, 2021-08-19:
I also need the drivers. please that's more than 4 people who've requested the driver... someone reupload it to dropbox or one drive..... we would all really appreciate it. @haakonstorm please help us out  @jziolkowski @mbspears @cwilmo

ViroAR, 2021-08-19:
Hi guys. The Support of FTDI sent me the .zip with drivers for ARM64.
You can download with this link: https://www.mediafire.com/file/6h60x6kmkufzd6c/CDM_v2.12.32.B3_for_ARM64.zip/file
[https://static.mediafire.com/images/filetype/download/zip.jpg]<https://www.mediafire.com/file/6h60x6kmkufzd6c/CDM_v2.12.32.B3_for_ARM64.zip/file>
CDM v2.12.32.B3 for ARM64<https://www.mediafire.com/file/6h60x6kmkufzd6c/CDM_v2.12.32.B3_for_ARM64.zip/file>
MediaFire is a simple to use free service that lets you put all your photos, documents, music, and video in a single place so you can access them anywhere and share them everywhere.
www.mediafire.com

Best regards!

ViroAR, 2021-08-19:
Hi guys @smokescreen42 @Noenemy88 @Alexandtr1 @M1LLZR . The Support of FTDI sent me the .zip with drivers for ARM64.
You can download with this link: https://www.mediafire.com/file/6h60x6kmkufzd6c/CDM_v2.12.32.B3_for_ARM64.zip/file

Best regards!

smokescreen42, 2021-08-19:
thank you so freakin much !!!!!

ViroAR, 2021-08-19:
Hope u can resolve it! @smokescreen42

smokescreen42, 2021-08-19:
this is me reporting back that it works and I was just able to get K Line synced under the com port and erase my codes !

AlexandruB30, 2021-08-22:
Hey @smokescreen42, 

How did you set it up on your M1? I've got a MacBook Air M1 with a FT232R running WIn10 on Parallels 17. I've tried installing the drivers using Test Mode but I keep getting this error:

[image]

Jason2866, 2022-03-15:
Using with Apple Silicon. Just install PyQt5 via Homebrew and it just works

closed 2022-03-15 by Jason2866

tokamac, 2023-05-15:
> Using with Apple Silicon. Just install PyQt5 via Homebrew and it just works

With everything installed and up-to-date regarding brew, pip3, python3… when I try to install from PyPI using pip on my iMac M1 with:
`pip3 install tasmotizer`

I get the error:
```Preparing metadata (pyproject.toml) ... error
 error: subprocess-exited-with-error
   × Preparing metadata (pyproject.toml) did not run successfully.
   │ exit code: 1
 ╰─> [22 lines of output]
      Traceback (most recent call last):
        File "/Users/USER/Library/Python/3.9/lib/python/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Users/USER/Library/Python/3.9/lib/python/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/Users/USER/Library/Python/3.9/lib/python/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/sipbuild/project.py", line 586, in setup
          self.apply_user_defaults(tool)
        File "project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/sipbuild/project.py", line 237, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/mm/pzbcq3sx7xncs5r7wbyscbvr0000gn/T/pip-build-env-xq9db0xz/overlay/lib/python3.9/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Jason2866, 2023-05-15:
Can not reproduce. M1 Air (2020) with latest Ventura
```
Last login: Mon May 15 14:54:36 on ttys162
hans@MacBook-Air-von-Johann ~ % pip3 install tasmotizer
Collecting tasmotizer
  Downloading tasmotizer-1.2.1-py3-none-any.whl (196 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 196.8/196.8 kB 5.0 MB/s eta 0:00:00
Requirement already satisfied: pyserial>=3.0 in /opt/homebrew/lib/python3.11/site-packages (from tasmotizer) (3.5)
Requirement already satisfied: PyQt5>=5.10 in /opt/homebrew/lib/python3.11/site-packages (from tasmotizer) (5.15.7)
Requirement already satisfied: PyQt5-sip<13,>=12.11 in /opt/homebrew/lib/python3.11/site-packages (from PyQt5>=5.10->tasmotizer) (12.11.0)
Installing collected packages: tasmotizer
Successfully installed tasmotizer-1.2.1
```

tokamac, 2023-05-15:
I was able to solve the problem by installing latest python (currently version 3.11.3) with HomeBrew, making it the default one (instead of Python 3.9 that comes with Xcode) and add its directory to my PATH. It works.

villasenor, 2023-12-11:
Just to add another data point - seems to be working fine for me (M1 Pro, Ventura) with just a `pip3 install tasmotizer`, and downloading `tasmotizer.py` and running `python3 tasmotizer.py`.