tasmota/tasmotizer #92: run into an error on MAX OS X BirSur 11.2.2

homefuchs, 2021-03-08:
Hi,

just installed Tasmotizer on my MAC but unfortunately it terminates with the following error message:
File "./tasmotizer.py", line 380
    backup_size = f'0x{2 ** self.backup_size}00000'
                                                  ^
SyntaxError: invalid syntax

What can I do?

BR,
Ulrich

jziolkowski, 2021-03-08:
Which installation method?
Which python version?

homefuchs, 2021-03-08:
Hi,

with python 3.8
and pip3 install tasmotizer

jziolkowski, 2021-03-08:
that's very odd. Do you have the same error when trying the app build from release?

fabiosoft, 2021-03-11:
Tested a release (last) version on Big Sur and it quits on boot.

Jason2866, 2021-03-11:
Please try with Python 3.9.x. Check pyserial version. Has to be v.3.5 (or greater)
PyQt5 version 5.14.2. is known working (later versions can make issues)
The compiled App is not working (i have removed)

wowulavega, 2021-11-08:
Hi,  
to make it easy, I wold like to attach to this thread.
My Tasmotizer also don’t install (macOS Big Sur 11.4 / Python 3.10.0)
After starting TS App. the Beachball appears and never goes away until I interrupt the installation.
[1108/150251.458798:WARNING:dns_config_service_posix.cc(335)] Failed to read DnsConfig.
[1108/162654.929023:ERROR:cache_util.cc(140)] Unable to move cache folder GPUCache to old_GPUCache_000
[1108/162654.929974:ERROR:disk_cache.cc(184)] Unable to create cache
[1108/162654.930167:ERROR:shader_disk_cache.cc(622)] Shader Cache Creation failed: -2

jziolkowski, 2021-11-08:
Doesn't install or doesn't run? Also I don't see anything related directly to Tasmotizer or PyQT in this error log, sorry

wowulavega, 2021-11-08:
…. doesn’t run.
The download installs to an app. If I open that app… the beachball appears and rolls until I stopp the app.

Jason2866, 2021-11-09:
As i wrote directly above you post. I have removed the Mac App since it does not work.
The only way afaik to get Tasmotizer working is installing the python variant.
No App!. Install the needed PyQt5 via Homebrew. It takes care to get a working version.

wowulavega, 2021-11-09:
Thanks for the hint … unfortunately I am not so familiar with using datelines, so I decided to skip the whole Tasmota option.

I just don't understand why people publish apps that don't work.

But thanks for you’re reply.

jziolkowski, 2021-11-09:
> I just don't understand why people publish apps that don't work.

You're more than welcome to provide a PR to make it work.

closed 2021-11-09 by jziolkowski

wowulavega, 2021-11-09:
The emphasis was on "publish".
After all, a car manufacturer doesn't put cars on the market that can't at least drive.

Jason2866, 2021-11-09:
You can compile apps for platforms you dont have and doing this because people friendly asked for. And if this app does not work (because people tested) it is removed.
As simple as this. But thanks for your advice, i will never do again.

fabiosoft, 2021-11-09:
So tasmotizer just lost all MacOS users... what a pity.
All people will migrate to other tools

jziolkowski, 2021-11-09:
@fabiosoft sponsor me a Mac, I will gladly make a working version for you.