tasmota/tasmotizer #99: cannot load bin data error

MinaMalak-cmd, 2021-04-16:
when I switched from 'BIN File' mode into 'release ' mode it gives me an error of  'cannot load bin data expecting value' as shown in image attached below

[image]

jziolkowski, 2021-04-16:
Hmm, does the release button show the version next to it? I assume that it's connected to internet...

MinaMalak-cmd, 2021-04-16:
@jziolkowski if i understand your question correctly, no it doesn't show any thing

jziolkowski, 2021-04-16:
can you paste screenshots of the interface with 'release' and then 'development' clicked?

MinaMalak-cmd, 2021-04-16:
I couldn't proceed to release every time i try it to click on development or release it gives me that error and stop on BIN file

jziolkowski, 2021-04-16:
ahh got it. so perhaps it's an external issue with the JSON data with builds information. I'll check it.

MinaMalak-cmd, 2021-04-16:
ok waiting for your checking, hope this happen soon

ascillato, 2021-04-16:
I have just tested and works fine in my PC.

[image]

Do you have internet connection on that computer? or an active firewall?

MinaMalak-cmd, 2021-04-16:
yes I have good internet connection, and disabled anti virus, is there any other thing to do to solve this problem?

MinaMalak-cmd, 2021-04-16:
the problem has been solved it was due to antivirus

closed 2021-04-16 by MinaMalak-cmd

ascillato, 2021-04-16:
I could reproduce your error. It is given only when Tasmotizer can't access internet.

@jziolkowski Is it possible to change the error message to something like **_A Internet connection is required for this feature to work_** ?