tasmota/tasmotizer #110: HOSTNAME AND FRIENDLY NAME not correct

scargill, 2021-08-16:
I'm suer I've brought this up before - but after flashing an ESP8266 with Tasmotizer then entering the SET CONFIG menu - I put in a name "dimmer3" without quotes - I've done similar many times and it is always wrong.

So now when I go looking for the device by name on the network I find a device dimmer3-7867 instead of dimmer3 - which kind of defeats the object. 

hostname is %s-%04d
Surely it would make sence to set the hostname in this case to dimmer3 in that config.
Under Tasmota menu OTHRE we find friendlyname is still tasmota and friendlyname1 is dimmer3 - so even that isn't set up usefully. Is there any chance you could add options to set all 3 to (in this case) dimmer3 in one go in the config instead of having to find the device with ADVANCED IP SCANNER then go into the Tasmota config and tidy this lot up. It would make life easier.

Also I'm, not sure I see the point in SEND CONFIG of the fixed backup IP? Personally I'd find it more useful if this was set to the second access poinr that Tasmota allows?

Great free software but this has been a long term issue for me...  worth a look?

sausin, 2021-09-02:
+1 wrt hostname

scargill, 2021-09-02:
> +1 wrt hostname

Sorry that makes no sense to me...

sausin, 2021-09-02:
I meant to indicate that I was also facing the same issue as you with respect to the hostname on network.

scargill, 2021-09-02:
> I meant to indicate that I was also facing the same issue as you with respect to the hostname on network.

Thanks -  few people have mentioned this to me - nothing from the author yet